karawin/Ka-Radio32 #112: No good reason for limiting memcpy size in spiram_fifo.c ?

har-in-air, 2019-04-19:
Not sure if there are legacy issues which resulted in the fifo read and write being limited to 1024 bytes at a time (some other 24LC type serial ram perhaps).  I cleaned up spiram_fifo.c and this code works fine for me. I have a espwrover homebrew board based internet radio and am using 220*1024 sized fifo without any issues with the attached code. I limited fifo size to 220kbytes because i never saw the buffer fill up to more than ~50% when I used 420*1024 as per original code 

(I have a slow wifi hotspot connection to internet).
[spiram_fifo.c.txt](https://github.com/karawin/Ka-Radio32/files/3097774/spiram_fifo.c.txt)

karawin, 2019-04-19:
To be generic, the ram is detected and if present, the size is 
#define BIGMEMORY 131072
This is about 30seconds of play without reception.
It is enough. More is disturbing the title detection. 
As you noticed, it is hard to fill the buffer more than that.
Your solution is good for you but, only for configuration with the external ram.
A clenest solution is to change the define in buffer.h.

But it not an issue for my point of view.
buffer.c:
ICACHE_FLASH_ATTR void initBuffer()
{
	 if (externram== false)
	 {
//		 BUFFER_SIZE = 12960;
		 BUFFER_SIZE = 14080;
//		 BUFFER_SIZE = 16000;
		 buffer = malloc(BUFFER_SIZE);
	 } else{
		 BUFFER_SIZE = BIGMEMORY;
	 }	 	 
}

ICACHE_FLASH_ATTR void initBuffer()
{
	 if (externram== false)
	 {
//		 BUFFER_SIZE = 12960;
		 BUFFER_SIZE = 14080;
//		 BUFFER_SIZE = 16000;
		 buffer = malloc(BUFFER_SIZE);
	 } else{
		 BUFFER_SIZE = BIGMEMORY;
	 }	 	 
}

har-in-air, 2019-04-19:
Its not the size of the fifo ram i'm commenting about, just that the fifo read and writes  break up the buffer into  1024 byte chunks in the existing code, and I don't see a good reason for that.

karawin, 2019-04-19:
Just to limit the time and let the hand to the scheduler.
the spi bus is slow and may be needed by the vs1053 and lcd if presents.

har-in-air, 2019-04-19:
OK thanks. I haven't integrated an LCD yet so did not see any issues with user interface responsiveness.

closed 2019-04-19 by har-in-air

karawin, 2019-04-19:
Ok. 
A generic software is not so easy to write ;-)
Thanks.