karawin/Ka-Radio32 #113: Missing command for Telnet

bazooka07, 2019-04-21:
Hi,
I'm playing with cli.instant command on telnet.

Can you add a telnet command for adding the url of the instant play at the end of the playlist ?
And send notification  to the web browser and update the #full_URL input field in the start page ?

karawin, 2019-04-23:
Bonne idée, mais difficile à réaliser. Il faut aussi réaligner les webs connectés.
Je vais déja ajouter un websocket  pour mettre à jour le champ "instant play"

bazooka07, 2019-04-23:
Si "instant play" se met à jour dans le navigateur, ce sera déjà un plus. On pourra faire un copier-coller manuel de la "full path vers la page d'édition de la liste des stations".
Après, on peut rajouter un bouton à coté du champ d'entrée de "instant play" pour automatiser le copier-coller en javascript. Cela sera plus facile à faire qu'en C++.

karawin, 2019-04-24:
oui, je vois. Je vais faire ça.

karawin, 2019-04-25:
A tester dans la prerelease
sys.prerelease

bazooka07, 2019-04-27:
J'ai fait quelques essais. Ca marche bien apparemment.
Quand on clique sur le bouton "Add & Edit", on se retrouve avec un champ "Name" vide.
C'est possible de le remplir avec la valeur de CLI.ICY0 ?

karawin, 2019-04-27:
Tout dépend quand on édite. 
Pas simple en tout cas. Il faut vérifier que c'est en lecture et que c'est la bonne radio et que ICY0 est valide.
Je ne vais pas m’embêter avec ça. Je préfère que ce soit manuel. (pour le moment....)

karawin, 2019-04-29:
Delivered....

closed 2019-05-01 by karawin