karawin/Ka-Radio32 #115: Crash if lcd.type is 255

pilnikov, 2019-05-03:
Hello JP. I have problem again

Part of Config file

[image]
 
Result

[image]

Switch from 192 to 255

[image]

I suggest to do a preliminary analysis of the config file and form several flags according to its results.
Example: O_LCD_TYPE, data, u8,255 -> Display is "Not used" -> a ban on launching all the blocks responsible for initializing the display and outputting information to it.
Example2: The default audio output type is I2S, but in configuration:
P_I2S_LRCK, data, u8,255
P_I2S_BCLK, data, u8,255
P_I2S_DATA, data, u8,255
The system will start swearing at the wrong configuration of the GPIO -> if I2S is prohibited in the config file - do not start its handler. By analogy, you can do with handlers of buttons,  joysticks, encoders and IR
tnx

karawin, 2019-05-03:
Yes. A problem.
update again. It works now.
A complete correction soon.

karawin, 2019-05-13:
Ok in R4

closed 2019-05-13 by karawin