karawin/Ka-Radio32 #116: SPI LCD Type support

WillVi1, 2019-05-03:
Hello, 

thank you for this awsome project!! 

I´ve some LCDs with ST7565 Controller (128x64). I tried the "LCD_SPI_ST7565_ZOLEN	66", and "LCD_SPI_ST7565_NHD_C12864	76", but the contrast is here very poor (not readable). 

I tried with an arduino uno the U8G2 Lib - same problem with "Zolen" and "NHD". But when I use the type "U8G2_ST7565_64128N_F_4W_SW_SPI u8g2" it works fine! 

So, I would be very pleased if you could add this U8G2 LCD Type to the LCD list.

karawin, 2019-05-04:
Ok
Will be 
#define LCD_SPI_ST7565_64128N	77 //128X64
Update and tell me if it works

WillVi1, 2019-05-04:
Hello, 
I don´t have the toolchain installed. 
Could you please add the define in the next revision?

karawin, 2019-05-04:
I made it in the bin.
Just update and type the new
sys.lcd("77")

WillVi1, 2019-05-04:
Hello, 
sorry, but I can´t find the bin. The last change of the github project is 19hours ago. 
Trying to use the type 77 crashed my radio (need to flash the bins with flash download tool.

karawin, 2019-05-04:
Update with the web page or sys.update command.
The release number is the szame.

WillVi1, 2019-05-04:
It´s working, thank your very much!

X-Stas-EEE, 2019-05-12:
Hello,
The ST7920 display works incorrectly - sys.lcd("73").
As I have figured out, it is related to SPI Clock for this controller: 1.25 MHz. You use 10 MHz.
I installed the toolchain and tried to compile the project. However, it was built with several warnings and ESP32 didn't start after flashing.
Could you, please, fix this issue?
Maybe try
dev_config.clock_speed_hz = u8x8->display_info->sck_clock_hz;
at u8g2_esp32_hal.c line73?
Thank you.

karawin, 2019-05-13:
Try the prerelease
sys.prerelease

X-Stas-EEE, 2019-05-13:
Thanks for the quick reply (as always)!
I'll try prerelease as soon as I get home!

X-Stas-EEE, 2019-05-13:
I have tried the prerelease and the picture looks much better now, but it still has some artifacts in the upper quarter of the screen. 
Experimentally, I found out it can work at 400 KHz (just assembled the build with warnings, but it could display the starting screen correctly).
Could you, please, try: 
dev_config.clock_speed_hz   = 400000;
at u8g2_esp32_hal.c line73?
Thanks!

karawin, 2019-05-13:
Try it.

X-Stas-EEE, 2019-05-14:
Finally, it works fine!
Thank you very much!

closed 2019-05-14 by karawin