karawin/Ka-Radio32 #119: encoder1 does not change the station

dimaw, 2019-05-29:
I have configuration: esp32s + vs1053 + encoder in "1" mode (the station control and volume change when pushed and held), without display. Flashed KaraDio32 V1.9 R4.
When I push and rotate volume is changed ok. But if I rotate only, stations not changes.
In my config file ([standard_adb.csv](https://github.com/karawin/Ka-Radio32/files/3233137/standard_adb.csv.txt)):

> ...
> P_ENC0_A,data,u8,255
> P_ENC0_B,data,u8,255
> P_ENC0_BTN,data,u8,255
> P_ENC1_A,data,u8,16
> P_ENC1_B,data,u8,17
> P_ENC1_BTN,data,u8,5
> ...
> O_LCD_TYPE,data,u8,255
> O_LCD_ROTA,data,u8,0
> O_LCD_OUT,data,u32,0
> O_DDMM_FLAG,data,u8,0
> ...

In logs all ok:
> 17:29:47.964 -> D (96492) addon: encoder value: 3
> 17:29:48.170 -> D (96692) addon: encoder value: 2
> 17:29:48.580 -> D (97092) addon: encoder value: 1
> 17:29:48.751 -> D (97292) addon: encoder value: 5
> 17:29:48.956 -> D (97492) addon: encoder value: 9
> 17:29:49.162 -> D (97692) addon: encoder value: 8
> 17:29:51.180 -> D (99692) addon: encoder value: 2
> 17:29:51.352 -> D (99892) addon: encoder value: 9
> 17:29:52.552 -> D (101092) addon: START/STOP State: 1
> 17:29:52.586 -> ##CLI.STOPPED# from addon stop
> 17:29:52.586 -> D (101092) vs1053: Decoder vs1053 stopped.
> 17:29:52.586 -> 
> 17:29:52.586 -> I (101092) vs1053: watermark: 198  408
> 17:29:52.586 -> D (101102) renderer: Renderer volume:  800
> 17:29:52.963 -> D (101502) renderer: Renderer volume:  3F00
> 17:29:52.998 -> I (101512) webclient: watermark : 45c  1116
> 17:29:52.998 -> I (101522) wifi: Set ps type: 2
> 17:29:52.998 -> 
> 17:29:58.850 -> D (107372) addon: encoder value: 2
> 17:29:59.058 -> D (107572) addon: encoder value: 4
> 17:30:00.254 -> D (108772) addon: encoder value: 11
> 17:30:00.458 -> D (108972) addon: encoder value: 11
> 17:30:00.630 -> D (109172) addon: encoder value: 19
> 17:30:00.836 -> D (109372) addon: encoder value: 24
> 17:30:01.041 -> D (109572) addon: encoder value: 27
> 17:30:01.250 -> D (109772) addon: encoder value: 30
> 17:30:01.454 -> D (109972) addon: encoder value: 40
> 17:30:02.001 -> D (110542) addon: encoder value: 36
> 17:30:03.032 -> D (111542) addon: encoder value: -11
> 17:30:03.205 -> D (111742) addon: encoder value: -13
> 17:30:03.411 -> D (111942) addon: encoder value: -13
> 17:30:03.616 -> D (112142) addon: encoder value: -11
> 17:30:03.823 -> D (112342) addon: encoder value: -15
> 17:30:09.829 -> D (118342) addon: START/STOP State: 0

nano111, 2019-05-29:
Did you use pullup resistors?

dimaw, 2019-05-29:
Yes, this is standard rotary encoder KY-040, resistors built in: https://ru.aliexpress.com/item/2PCS-FREE-shipping-Rotary-Encoder-Module-Brick-Sensor-Development-for-Dropshipping-KY-040/32474584136.html

nano111, 2019-05-29:
Strange, if volume works the stations should also.No ideas,sorry.

karawin, 2019-05-29:
What  is the software version used?
Do you have a lcd?
the encoder change the station but the change is taken into account in 3 sec to let the user changes again.
It is better with a lcd to show the station chosen before the start of the play;

dimaw, 2019-05-30:
Can I change this timeout without recompiling? Maybe 1 sec will better for me (without display)?

karawin, 2019-05-30:
Please turn the encoder one notch only and see what happen.
The value must be 1 or -1

11:57.8: D (872747) addon: encoder value: 1
12:00.0: D (874997) decoder: Decoder stopped.

D (874997) decoder: MAD decoder stack: 496

12:01.0: I (875987) wifi: Set ps type: 2

##CLI.NAMESET#: 26 RFM Thematique 2
##CLI.URLSET#: rfm-wr14-mp3-128.scdn.arkena.com
##CLI.PATHSET#: /rfm.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
12:01.3: I (876277) wifi: Set ps type: 2

12:01.4: I (876287) wifi: Set ps type: 1

##CLI.VOL#: 41
D (876287) renderer: Renderer volume:  800
I (876287) webserver: playstationInt: 26, g_device: 25
D (876287) eeprom: saveDeviceSettings
12:01.9: D (876707) renderer: Renderer volume:  800
12:01.9: I (876717) webclient: watermark : 3bc  956
I (876717) webclient: Webclient socket: 61, errno: 0
12:02.5: D (877327) webclient: contentType: 5
12:02.6: ##CLI.ICY0#:  RFM Thematique 2
12:02.7: ##CLI.ICY3#:  http:\/\/www.rfm.fr
12:02.8: ##CLI.ICY4#:  Radio
12:02.9: ##CLI.ICY5#:  128
12:03.0: ##CLI.ICY6#:  RFM Thematique 2
12:03.1: ##CLI.ICY7#:  bitrate=128
D (877927) webclient: t1: 0x3ffc3078, cstatus: 2, icyfound: 1  metad:16000 Metaint:16000

D (877927) webclient: newlen: 844   len: 1436   chunked:0  pdata:3ffc2e2c
12:03.3: D (878127) renderer: Renderer volume:  800
##CLI.PLAYING#

One second or 3 or ten or anything else, without display it is unusable without the display. You cannot be sure of the station you select.
Use the android application or the web interface.

dimaw, 2019-05-30:
Understood thanks. In the terminal I set the type of display "0":
> sys.lcd("0")

and the encoder is now working in mode "1" correctly.

Please provide in the next version the ability to change the timeout in the configuration file. This is useful for those who do not have a display.

karawin, 2019-05-30:
Right. When type is 255, it doesn't work.
I will modify the code to allow it to work.
Thanks.

closed 2019-05-30 by karawin