karawin/Ka-Radio32 #120: I2s dac

nano111, 2019-05-29:
HI,
I have this dac:
https://fr.aliexpress.com/item/Max98357-I2S-3-W-classe-D-amplificateur-Interface-de-sortie-Dac-d-codeur-Module-carte-Audio/32988732683.html?transAbTest=ae803_3&ws_ab_test=searchweb0_0%2Csearchweb201602_9_10065_10068_319_10059_10884_317_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_10820_10301_10821_10303_537_536%2Csearchweb201603_52%2CppcSwitch_0&algo_pvid=f662d53f-7bc9-4179-9987-b51145e9eeb9&algo_expid=f662d53f-7bc9-4179-9987-b51145e9eeb9-2

connected to ttgo ts board LRCK BCLK and DIN and it works but with max volume.
When I lower volume to 0 ,sound stops,but just at 5 it goes to max level and I can not control it.
Any help?
thanks

karawin, 2019-05-29:
You must be in pdm mode on the board. Please select the pcm mode on the board and on the web page of KaRadio.

karawin, 2019-05-29:
Read that?
SD / MODE
This pin is used for shutdown mode but is also used for setting which channel is output. It's a little confusing but
essentially:
If SD is connected to ground directly (voltage is under 0.16V) then the amp is shut down
If the voltage on SD is between 0.16V and 0.77V then the output is (Left + Right)/2, that is the stereo average.
If the voltage on SD is between 0.77V and 1.4V then the output is just the Right channel
If the voltage on SD is higher than 1.4V then the output is the Left channel.
This is compounded by an internal 100K pulldown resistor on SD so you need to use a pullup resistor on SD to balance
out the 100K internal pulldown.
For the breakout board, there's a 1Mohm resistor from SD to Vin which, when powering from 5V will give you the
'stereo average' output. If you want left or right channel only, or if you are powering from non-5V power, you may need
to experiment with different resistors to get the desired voltage on SD

nano111, 2019-05-30:
Sure i set I2s on web page of the radio.Tried 2db gain,but very little difference.Little les volum ,but still no control.

nano111, 2019-05-30:
3db sorry

karawin, 2019-05-30:
Sorry, i don't have an answer for this board.

nano111, 2019-06-01:
It is confusing .I set PDM ,not I2S mode on web of radio and now it works Ok.

karawin, 2019-06-01:
Your board is a pcm/pdm input compatible but it seems that the pdm mode is better.
I had read a forum where a user of this board had a problem of volume. After changing to the same one, it was working ok.

nano111, 2019-06-02:
We learn every day :-)
I did try pdm before I wrote this issue , but it did not work because I did not hard reset the board.
Thanks

Pozdrav
Goran

nano111, 2019-07-06:
Just for info...
With above dac and karadio set to PDM the music is a bit louder and it plays slower.
When karadio is set to I2S (dont ask me how it started to work,it just did :-) ),speed is OK,but the volume is lower for about 1db.

closed 2019-07-08 by nano111

JayB7076, 2022-07-15:
Mine one is not even working any idea?
Any library or pin did I need to change?
I used Esp32 30pin Wroom board.
And I have same dac.

JayB7076, 2022-07-15:
I don't know why but after seeing photos of other people's project it seems like very few people used this type of dac....

nano111, 2022-07-16:
You have to tell us how did you connect wroom with dac.
In the configuration file you have to change pin numbers to match your connection.

JayB7076, 2022-07-21:
I connected same as shown in diagram.
And I didn't change any pin configuration from bin file.
Is there any way to check dac?

nano111, 2022-07-21:
Which diagram? Your connection needs to match with your csv file.
https://github.com/karawin/Ka-Radio32/tree/master/boards
Look here if there is no your board
you have to change pin numbers to match your connection. See instructions what to do with the csv file you modified.

nano111, 2022-07-21:
I do not know the way to test the dac.

JayB7076, 2022-07-21:
I connected according to Karadio32 diagram
Diagram is in Karadio32_master>images>schematic.
And I used Esp32 30pin Wroom board and I have same DAC as you showed in AliExpress link.

nano111, 2022-07-21:
I think you do.Tell me numbers of io pins wher you connected pins Din, bclk, sd and lrc on your wroom board.
Do you have a display or rotary encode or anything other than wroom and dac connected?

nano111, 2022-07-21:
…or send mw exact link to your diagram.

JayB7076, 2022-07-22:
[image]
[image]
[image]

Yes I connected display and rotary encoder but this two is directly connected to Esp32.
I2S DAC --> ESP32
LRC --> D25
BCLK --> D26
DIN --> D22
GAIN IS NOT CONNECTED ANYWHERE
SD IS NOT CONNECTED ANYWHERE
GND --> ESP32 GND 
VCC --> VIN OF ESP32

nano111, 2022-07-22:
“ standard_psram.bin is the default description for a wrover cpu”

check if this csv file in boards directory matches your connection.
Then make
bin with that file as described.

JayB7076, 2022-07-22:
No I didn't use standard_psram.bin file and before uploading karadio32 bin files I formatted flash of esp32 board.and I uploaded karadio32 default factory flash but I don't know it's factory one or not.
And I used Esp32 Wroom board and it have 30pin not wrover.

nano111, 2022-07-22:
as I said … you have to modify csv file according to your connection and 
convert it to bin … then flash it tothe board

JayB7076, 2022-07-23:
I connected pins according to bin file and in first time uploading bin files I didn't find any standard_adb.bin file in Karadio32_master
[image]
Than I found file and re-uploaded five bin files again.
[image]
[image]
And I connected again according to standard_adb.bin and I didn't changed anything I uploaded bin file that i found in karadio32_master.

JayB7076, 2022-07-23:
Can't I just used already made bin file?
And did standard_adb.bin file is not for Esp32 30p Wroom?
And is it compulsory to change CSV file and than convert to bin and than upload?

nano111, 2022-07-23:
it is not necessary to change csv if everything is connected as in that file.
Did you change in web interface of karadio in Setting audio option to I2S or PDM?
Does your display work?Encoder?
You can try to change pins 25 and 26 anf connect da. to some other two and change csv.Maybe those are only for internal dac on your board.
Maybe you have a bad dac board if nothing of this helps

JayB7076, 2022-07-23:
Yes I connected everything according to CSV file.
Yes I tried every options from karadio32 web interface and non of them worked accept Dac option,and my internal DAC is working completely fine.
Yes my display and rotary encoder is working fine.
Yes and only 25 and 26 is only two pins for internal DAC.
I am thinking to buy and try new dac but this time VS1056 instead of max98357a.
Is this standard_psram.bin file can work on esp32 30p Wroom?

JayB7076, 2022-07-23:
Can it supports VS1003b module?
I found one it's similar look similar to vs1053
And some people used it so can it going to work?

nano111, 2022-07-24:
Try to connect lrck and bclk to two other io pins that are free and modify csv to that connection and make bin.
Do you hard reset (disconect power to the board) your board after  you change audio settings on web interface?
You always have to check csv file if it matches your connection .

JayB7076, 2022-07-24:
Yes I hard reseted board after changing settings from url.
And CSV file connection matches to..

JayB7076, 2022-07-24:
I gess I need to try new one.
I found total 3 boards that can delever fast to my address and seller Shop are in my country and price is looks ok too..
But I am confused which to use
Any recommendations?
Mine not working dac is max98357a 
And I found Sparkfun one but what if this not work?
[image]
And I found this vs1003b but on board they printed vs1003b/vs1053 so I am confused but ic number is vs1003b and I seen a someone used before but he programmed esp with Arduinoide and code is not Karadio one.
[image]
And I found this vs1053 in Arduino shield board type but still not used by anyone so don't know if it going to work but ic is vs1053 so thinking to use.
[image]

JayB7076, 2022-07-24:
And do you know how do I add url of radio station?
I added some but they are not working.

nano111, 2022-07-24:
I used VS1053 in one  of my karadio and it works perfectly.For other I do not know.
Is your dac new or used?
Did you try using other gpio’s insted of 25 and 26?
I think you are missing something.
That dac works well in my radio.

nano111, 2022-07-24:
> And do you know how do I add url of radio station?
> I added some but they are not working.

How did you try yor dac if you didnt add stations to your radio?
There are some radio lists for download on radios web interface under settings scroll to the bottom.
Upload one of them in your radio.
You can edit list under Edit menu … click on station and you can change url and name. You have to check if the stream is in mp3 format.

JayB7076, 2022-07-24:
No no I already added list of radio station but the list is from karadio32 I am not added my country's station.
Yes I changed pins but still no sound only internal dac is working.

nano111, 2022-07-24:
“ For a wrover cpu you need a csv file with psram in the name. Without it, the default configuration will fail.”
That is quote from karadi32 read me.
Maybe you could try that , maybe the boards are similar wroom and wrover.

nano111, 2022-07-24:
https://swling.com/blog/2021/04/robs-tips-for-uncovering-radio-station-stream-urls/

here are instructions for finding urls of web radio stations

nano111, 2022-07-24:
when yoy chsnged pins did you modified csv file and made a new bin
and flash it to the board?

JayB7076, 2022-07-25:
Yes I modified pins after seeing pin diagram of esp32
And flashed to Esp32.
And I found how to find radio station url.

JayB7076, 2022-07-25:
Hey did Karadio32 supports m3u8 codec?
I found this link but it seems like it's streaming in m3u8 format

nano111, 2022-07-25:
https://online-radio.eu/radio/92592-radio-mirchi-patna
On this page you can download 
list for vlc, winamp (in green window)
Paste url from the list to your radio… it should work

JayB7076, 2022-07-25:
"No ip found" this error massage showing

JayB7076, 2022-07-26:
Can I use that red colour vs1053 board?
What's you opinion?

nano111, 2022-07-26:
VS1053 is Ok… but if your max is new I do not see reason that it wont work.
You are making  a mistake somewhere and you should try to find it.
Abiuy urls see tutorial that I sent you the link to.

JayB7076, 2022-07-26:
I have one question after uploading bin file to Esp32 you need to format it's flash for new bin file upload right?but after formatting you need to upload factory flash file right?I once formatted Esp32 and uploaded factory bin file is it right thing to do?

JayB7076, 2022-07-26:
And there is bigger problem occurs with me my pc is have gone for troubleshooting I used my friends pc for programing Esp32 and now his pc have some problem so I am stuck I can't change program only I can do is change everything from site....

nano111, 2022-07-26:
WARNING: the standard_adb.bin must be changed for a wrover processor or another configuration.
 Do you use this tool?
http://karadio.karawin.fr/karawin32Flash.jpg
You only need to flash your radio once and after just that last file if you change your hardware configuration.

JayB7076, 2022-07-27:
Yes I used Esp32 flash tool.
And I uploaded karadio32-Karadio32-partition-bootloader-standerd_adb BIN files
I used Esp32 Wroom 30pin.

nano111, 2022-07-27:
Maybe you could try  that psram file for wrover.

JayB7076, 2022-07-28:
Man I don't have PC for now so.... waiting...
And I ordered vs1053 so I am going to try that to..
And whenever I have my pc back i will upload param.bin file for sure.

JayB7076, 2022-07-29:
[image]
Yep you are right I made mistake in program somewhere.
I connected this vs1053b and while radio is playing an led indicator light blinks but no sound.

JayB7076, 2022-07-29:
But how I uploaded 5 required bin files.
Karadio32
Karadio32
Partition
Bootloader
Standard_adb
And I didn't change anything from it and pins are right to.

JayB7076, 2022-07-29:
I gess I need to upload it again

JayB7076, 2022-07-29:
Oooh now I realise that..
WARNING: the standard_adb.bin must be changed for a wrover processor or another configuration.
I need to change even I have same pins as standard_adb.
And there is no Wroom in -- WARNING: the standard_adb.bin must be changed for a wrover processor or another configuration. -- so I didn't changed anything.

nano111, 2022-07-29:
does it work?

JayB7076, 2022-07-30:
Na still not uploaded code but did I need to format Esp32 right?for new bin files?

nano111, 2022-07-30:
Why dont you read what I wrote?

> WARNING: the standard_adb.bin must be changed for a wrover processor or another configuration.
> Do you use this tool?
> http://karadio.karawin.fr/karawin32Flash.jpg
> You only need to flash your radio once and after just that last file if you change your hardware configuration.

I do not know how you format your esp32 just turn off first 3 files inflash tool
karadio32-Karadio32-partition-bootloader and leave last one which you made from your configuration or 
use alredy made psram bin .You okly need to flash everything frat time…
then you just change last bin which is hardware configuration, try if it works, if not change configuration and flash again with  new file.

JayB7076, 2022-07-30:
But how to turn off files from Esp32 tool?

nano111, 2022-07-30:
Just uncheck first 3 files on the left side of the file location in tool.

JayB7076, 2022-08-05:
No i mean I uploaded bin files on esp32 so I need to format first? Or its going to replace bin files like Arduino?

JayB7076, 2022-08-21:
Upload code again still not working
Which bin file are I need to upload for Esp32 Wroom 32p?

[4 more replies]