karawin/Ka-Radio32 #121: Error when building

bazooka07, 2019-06-08:
Hi,
I'm getting the following error when I'm trying to build :

> /home/jpierre/esp/Ka-Radio32/main/./app_main.c: In function 'app_main':
> /home/jpierre/esp/Ka-Radio32/main/./app_main.c:982:2: error: 'else' without a previous 'if'
>   else
>   ^
> /home/jpierre/esp/esp-idf/make/component_wrapper.mk:273: recipe for target 'app_main.o' failed
> make[1]: *** [app_main.o] Error 1

Any idea for fixing ?

bazooka07, 2019-06-09:
I proceed a new install of esp-idf and Ka-radio32 on Ubuntu 18-04.
For esp-idf I had following instructions here :
https://docs.espressif.com/projects/esp-idf/en/v3.2/get-started/linux-setup.html
And that builds right.
Please add this link in README.md

Sorry for this trouble.

karawin, 2019-06-10:
Hi JP,
I can't explain why the compiler failed on this line.
It is sure that a right installation of esp-idf is mandatory.

closed 2020-02-12 by karawin