karawin/Ka-Radio32 #124: WiFi password length

alexandertischenko, 2019-06-26:
After setting up the network through the AP mode, the device reboots and works normally. But in the settings menu, the "AP SSID:" and "AP PASSWORD:" all fields become empty. Perhaps the reason for this is the length of my wifi password of 17 characters?

karawin, 2019-06-27:
Please update again even if it is the same.
Some time ago i made a little change in this part of code.
NB: the password is 64 chars length.

alexandertischenko, 2019-06-28:
SSID and password is not displayed anyway. Probably the whole problem is in the special characters of my wireless password: «Gerri?hoEW<AP"0_»?

karawin, 2019-06-29:
Yes, the " is the problem. I will modify the code soon. Thanks

alexandertischenko, 2019-07-04:
After installing 1.9R5 WiFi password with character “ shows into the web interface AP PASSWORD field, but esp32 can’t connect with WiFi router and creates access point by default.

karawin, 2019-07-04:
Yes sorry a mistake.
Update again.

alexandertischenko, 2019-07-05:
Works fine. Thanks!

closed 2019-07-05 by alexandertischenko