karawin/Ka-Radio32 #126: Wrong time zone

Valenttt, 2019-07-02:
Hi,
When setting the time zone +10 time is displayed incorrectly, it adds 1 hour.
There is no transfer of hours for summer time in Russia.

karawin, 2019-07-02:
The NTP server likewise has no knowledge of daylight saving time.

Valenttt, 2019-07-02:
I am now setting a time zone of +9 to get a time zone of +10, 
I hope the bug will be fixed.

Valenttt, 2019-07-03:
For comparison https://www.timeserver.ru/cities/ru/vladivostok

karawin, 2019-07-03:
The time zone is used to adjust the date and time with respect to universal time. An NTP server does not have the knowledge of summer hours. He does not know the Daylight Savings Time.
It is not a bug but a feature ;-).
May be this information can be found elsewhere on internet but I had not written such a code.

Valenttt, 2019-07-03:
ok

closed 2019-07-03 by Valenttt