karawin/Ka-Radio32 #128: LCD backlight

Valenttt, 2019-07-11:
Hi, I can’t implement the scheme for automatically turning off the backlight of the LCD display via sys.lcdout ("30"). 

[image]. 

In the file pattern.csv, GPIO22 cleaned and applied it for highlighting.

P_I2S_DATA, data, u8,255
P_BACKLIGHT, data, u8,22

then made pattern.bin, flashed to the address 0x3a2000. After the overload, the backlight turned on, but after 30 seconds. does not turn off.

Tell me what I'm doing wrong?

pilnikov, 2019-07-11:
Try this schematic diagram
[image]
102R = 1KOhm
103R = 10kOhm
Q1 - any n- channel mosfet, e.g. 2N7002

pilnikov, 2019-07-11:
в исходной схеме неверно включен полевик - он всегда открыт через паразитный диод, который есть физически, но на схеме не указан.

closed 2019-07-12 by Valenttt