karawin/Ka-Radio32 #131: amp control

petanque, 2019-08-12:
Im using a PAM8403 as amp and it work great except when there is no audio I can hear a small buzz.
I have tried esp-radio and it got a shutdown pin to control it, does Karadio got something like that? a pin that go high/low when there is no audio?

karawin, 2019-08-12:
Class D amp are very sensitive to noises.
Try to filter or add audio transformers
https://othermod.com/raspberry-pi-zero-audio-circuit/
https://fr.grandado.com/products/vodool-ground-loop-noise-isolator-audio-anti-jamming-device-noise-isolator-with-3-5mm-audio-interface-for-home-stereo-l3fe?variant=13650164613235&currency=EUR&gclid=CjwKCAjwnMTqBRAzEiwAEF3ndmhbFacTEIcpFXKXyHRjNpqmc45bbenfLHpBcCJb66e5TqEBMEM-sxoCPhUQAvD_BwE

mfdonmez, 2020-12-20:
Siz ingilizceye çevirin lütfen. 
Pam8403 üzerinde 8.bacak vref . 
Vs1053 den gelen gnd bacak 47 uf kapasitör ile pam8403 8. Bacak vref e bağla. Tahtadan ayırmadan üzerine tuttur. 
pam8403 gnd yi boş bırak. 
Ses problemi çözülüyor.