karawin/Ka-Radio32 #134: 1.8 inch SPI ST7735B LCD problem

Sera26, 2019-08-27:
Hello,

I have built another Karadio32 but with an ST7735B 128x160 LCD now. Everything works fine but LCD is turning to bright white after a few seconds. Reset of system can solve this issue temporarily but happens again after a while. Parallel to this problem picture/characters are falling apart randomly.
I'm using LCD type "192" because this was the best option up to now.
Can you help me please how to use this LCD correctly?

Sera26, 2019-08-28:
Connections are as follows:

LCD RST ---> ESP 32 IO2
LCD CS ---> ESP 32 IO13
LCD D/C ---> ESP 32 IO14
LCD DIN ---> ESP 32 IO23
LCD CLK ---> ESP 32 IO18
LCD VCC ---> ESP 32 +5V
LCD BL ---> ESP 32 +5V
LCD GND ---> ESP 32 GND

Are these pins linked correctly?

Jambola77, 2021-02-11:
Please check if 5V are ok for your display. Pins are ok. 
Then do a sys.lcd("192") and reboot.

closed 2021-02-21 by karawin