karawin/Ka-Radio32 #135: help with platformio build

brunnels, 2019-08-28:
I have it mostly building but I'm missing something.  Anyone interested in helping me to get it working?

First problem is 
```
main\websocket.c:286:57: error: 'header.maskKey' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                payload[i] = (payload[i] ^ header.maskKey[i % 4]);
```
I can get past that by commenting out the line since I'm just trying to get it to build with platformio right now.

It gets to the end and chokes on u8glib when linking.  
```
Archiving .pio\build\esp32doit-devkit-v1\libwpa_supplicant.a
Linking .pio\build\esp32doit-devkit-v1\firmware.elf
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.DrawString+0xc): undefined reference to `ucg_DrawString'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.DrawString+0x10): undefined reference to `u8g2_DrawUTF8'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.in_welcome+0x4): undefined reference to `ucg_SetColor'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.in_welcome+0x8): undefined reference to `u8g2_SetDrawColor'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.in_welcome+0xc): undefined reference to `ucg_DrawBox'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.in_welcome+0x10): undefined reference to `u8g2_DrawBox'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.lcd_welcome+0x10): undefined reference to `ucg_ClearScreen'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.lcd_welcome+0x14): undefined reference to `u8g2_ClearBuffer'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.lcd_welcome+0x18): undefined reference to `u8g2_FirstPage'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.lcd_welcome+0x1c): undefined reference to `u8g2_NextPage'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.task_lcd+0x2c): undefined reference to `u8g2_ClearDisplay'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.touchLoop+0x8): undefined reference to `haveTouch'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.touchLoop+0xc): undefined reference to `xpt_read_touch'
.pio\build\esp32doit-devkit-v1\src\addon.o:(.literal.task_addon+0x3c): undefined reference to `getTaskLcd'
```
plus 100 additional yellow lines and then the last few lines before error
```
.pio\build\esp32doit-devkit-v1\src\webserver.o:(.literal.setVolumei+0x0): undefined reference to `renderer_volume'
.pio\build\esp32doit-devkit-v1\src\webserver.o: In function `setVolumei':
webserver.c:(.text.setVolumei+0x3a): undefined reference to `renderer_volume'
.pio\build\esp32doit-devkit-v1\src\webserver.o: In function `setVolume':
webserver.c:(.text.setVolume+0x50): undefined reference to `renderer_volume'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32doit-devkit-v1\firmware.elf] Error 1
```

I probably also need the resulting sdkconfig.h from a regular Makefile build to replace the platformio generated one at main/sdkconfig.h

Here's platformio.ini I have so far:

```
[platformio]
src_dir = main
include_dir = main/include

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf
monitor_speed = 115200
monitor_port = COM3
build_flags =
    -Icomponents/audio_player/include
    -Icomponents/audio_renderer/include
    -Icomponents/bt_speaker/include
    -Icomponents/common/include
    -Icomponents/fdk-aac_decoder/include
    -Icomponents/fifo/include
    -Icomponents/MersusAudio/include
    -Icomponents/mp3_decoder/include
    -Icomponents/u8g2/cppsrc
    -Icomponents/u8g2/csrc
    -Icomponents/ucglib/cppsrc
    -Icomponents/ucglib/cppsrc_sdspi
    -Icomponents/ucglib/csrc
    -Icomponents/xpt2046
    -Icomponents/mad
```

SinglWolf, 2019-12-11:
HI!
If the problem is still relevant, I can help.
First you need to bring the directories of all resources to a format that is understandable to the PlatformIO.
E.g.:
```
|--components
|  |
|  |--audio_player
|  |  |--src
|  |     |- audio_player.c
|  |     |- audio_player.h
```
Then configure the PlatformIO, e.g.
```
[platformio]
src_dir = main
include_dir = main/include
lib_dir = components
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
monitor_speed = 115200
board_build.partitions = partitions.csv
build_flags =
    -Imain/include
	-Icomponents/audio_player/src
	-Icomponents/audio_renderer/src
	-Icomponents/common/src
	-Icomponents/fifo/src
	-Icomponents/mad/src
	-Icomponents/MerusAudio/src
	-Icomponents/u8g2/src
	-Icomponents/ucglib/src
	-Icomponents/xpt2046/src
```
Be sure to correct the error in the file partitions.csv, which is located in the root of the project !!!
It was
```# the stations partition
stations, 0x41,	     ,	 ,	       0X10000
```
Has become
```# the stations partition
stations, 0x41,	     ,	 ,	       0x10000
```
(I left only VS1053 support in the project. There may be a problem in the fdk-aac-oreo-m8 library. She has many nested directories.)
Have a good compilation !!! 😜

enjoyneering, 2020-04-15:
Hi, SinglWolf & brunnels

What version of IDF do you use? I've tried latest Platformio Espressif 1.12.0 (I believe it based on latest esp-idf 4.0) and it doesn't want to compile Ka-Radio32 & KaRadio32-pio.

SinglWolf, 2020-04-16:
> Hi, SinglWolf & brunnels
> 
> What version of IDF do you use? I've tried latest Platformio Espressif 1.12.0 (I believe it based on latest esp-idf 4.0) and it doesn't want to compile Ka-Radio32 & KaRadio32-pio.

Hello!
Before the last update, everything work fine with me. Now the same as yours :-)). I will ask questions to the developers PlatformIO.

enjoyneering, 2020-04-16:
Good. Thank you. I'm new to Platformo,

enjoyneering, 2020-04-16:
Hi SinglWolf,

Another question for you, most of the variables declared as "unsigned", not "unsigned int" or "unsigned bite" etc. What dose it mean? Checked all types.h in IDF, no info about just "unsigned".

Another dumb question why pointer "fakespiram" for "malloc" is "static char". Is it going to overflow because "char" is it 8 bit signed number from -128 to 127 & we reserve with "malloc"  420KB=420*1024 = 430080 bites?

Do you have another way to contact you? So I do not flood the thread with stupid questions?

SinglWolf, 2020-04-16:
> Another question for you, most of the variables declared as "unsigned", not "unsigned int" or "unsigned byte" etc. What dose it mean? Checked all types.h in IDF, no info about just "unsigned".

This does not happen. Give a specific example of where such an announcement occurs.

> Another dumb question why pointer "fakespiram" for "malloc" is "static char". Is it going to overflow because "char" is it 8 bit signed number from -128 to 127 & we reserve with "malloc" 420KB=420*1024 = 430080 bytes? How we are going to access last 430080 byte if the r/w pointer can go only up to 127? I'm I missing something?

A pointer cannot have a negative value by definition. In your example, the char is used as a byte, the number of bits is 8, which corresponds to a byte.

> Do you have another way to contact you? So I do not flood the thread with stupid questions?

I do not give lessons C. I will no longer answer such questions in this topic. If you want consult, write singlwolf@gmail.com

enjoyneering, 2020-04-16:
> This does not happen. Give a specific example of where such an announcement occurs

For example in  - Ka-Radio32/components/fifo/spiram_fifo.c

`static unsigned fifoRpos;`

> A pointer cannot have a negative value by definition. In your example, the char is used as a byte, the number of bits is 8, which corresponds to a byte.

well in this case it is a mistake in - Ka-Radio32/components/fifo/spiram_fifo.c

`static  char *fakespiram;`

and later in the code

`fakespiram = malloc(SPIRAMSIZE);`
...
`memset( fakespiram, 0, SPIRAMSIZE );`

> I do not give lessons C.

To bad, any way thank you for help and time.

SinglWolf, 2020-04-16:
> static unsigned fifoRpos;

"fifoRpos" doesn't tell you anything?

> well in this case it is a mistake in - Ka-Radio32/components/fifo/spiram_fifo.c

There are no errors, otherwise the compiler would not allow to build the firmware :-)
Understanding what pointers are in the C language is a very difficult task.
Any attempt to explain to You through Google’s translator will confuse You even more.