karawin/Ka-Radio32 #142: Compile Error

nopnop2002, 2019-10-29:
Hello.
I tried clean install.

My environment:

	ESP-IDF version:v4.1-dev-369-g4dac7c7df-dirty

My operation:

	$ rm sdkconfig

	$ make clean

	$make

	In file included from /home/nop/Ka-Radio32/components/audio_renderer/audio_renderer.c:17:
	/home/nop/esp-idf/components/driver/include/driver/dac.h:44:53: error: unknown type name 'gpio_num_t'
	 esp_err_t dac_pad_get_io_num(dac_channel_t channel, gpio_num_t *gpio_num);
	                                                     ^~~~~~~~~~
	/home/nop/Ka-Radio32/components/audio_renderer/audio_renderer.c: In function 'init_i2s':
	/home/nop/Ka-Radio32/components/audio_renderer/audio_renderer.c:102:3: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
	   if (i2s_driver_install(config->i2s_num, &i2s_config, 0, NULL) != ESP_OK)
	   ^~
	/home/nop/Ka-Radio32/components/audio_renderer/audio_renderer.c:104:4: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
	    if (i2s_driver_install(config->i2s_num, &i2s_config, 0, NULL) != ESP_OK)
	    ^~
	cc1: some warnings being treated as errors
	/home/nop/esp-idf/make/component_wrapper.mk:291: recipe for target 'audio_renderer.o' failed
	make[1]: *** [audio_renderer.o] Error 1
	/home/nop/esp-idf/make/project.mk:604: recipe for target 'component-audio_renderer-build' failed
	make: *** [component-audio_renderer-build] Error 2

pilnikov, 2019-10-29:
ESP-IDF version:v4.1 is not supported. Please use ver. 3.3 (Stable)

pilnikov, 2019-10-29:
https://github.com/espressif/esp-idf/issues/3938

nopnop2002, 2019-10-29:
Hello.
There are two problems during clean build

1.Kconfig.projbuild is not used.

2.partitions.csv does not take effect.
Must be specified in sdkconfig.defaults like this:

	CONFIG_PARTITION_TABLE_CUSTOM=y
	CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
	CONFIG_PARTITION_TABLE_CUSTOM_APP_BIN_OFFSET=0x10000
	CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
	CONFIG_APP_OFFSET=0x10000

pilnikov, 2019-10-30:
try my mod https://github.com/pilnikov/adf/tree/master/esp-adf/examples/Ka_mod

nopnop2002, 2019-10-30:
It become error:

	In file included from /home/nop/adf/esp-adf/examples/Ka_mod/main/include/webserver.h:13,
	                 from /home/nop/adf/esp-adf/examples/Ka_mod/main/app_main.c:78:
	/home/nop/adf/esp-adf/examples/Ka_mod/main/include/websocket.h:17:10: fatal error: crypto/sha1_i.h: No such file or directory
	 #include "crypto/sha1_i.h"
	          ^~~~~~~~~~~~~~~~~
	compilation terminated.
	/home/nop/esp-idf/make/component_wrapper.mk:291: recipe for target 'app_main.o' failed
	make[1]: *** [app_main.o] Error 1
	/home/nop/esp-idf/make/project.mk:604: recipe for target 'component-main-build' failed
	make: *** [component-main-build] Error 2

pilnikov, 2019-10-31:
You must replace the https://github.com/pilnikov/adf/tree/master/esp-adf/examples/Ka_mod/for_idf_replace/esp-idf/components/wpa_supplicant

karawin, 2020-02-07:
Try the SDK 3.3.1 release. It is the one used here.

nopnop2002, 2020-02-08:
Thank you.
But SDK 3.3.1 is berry bugy.

karawin, 2020-02-10:
Right but

nopnop2002, 2020-02-10:
I expect a port to 4.x.
Close this issue once.

closed 2020-02-10 by nopnop2002