karawin/Ka-Radio32 #144: Stream URL that doesn't work

bazooka07, 2019-12-28:
I wish to listen the stream from a web station with url as "http://5.135.154.69:12500/;"

If I am using curl as instantPlay() function in script.js with POST method, that's works :

`curl -d 'url=5.135.154.69' -d 'port=12500' -d 'path=/;' http://192.168.58.77/instant_play`

But if I use GET method like the instant command in https://github.com/karawin/Ka-Radio32/wiki, it fails :

```
curl 'http://192.168.58.77?instant=http://5.135.154.69:12500/;'
curl 'http://192.168.58.77?instant=http%3A%2F%2F5.135.154.69%3A12500%2F%3B'
```
The last form uses URL encoding but ka-radio does not seem supporting url encoding.

bazooka07, 2019-12-29:
From Shoutcast I get a link for http://yp.shoutcast.com/sbin/tunein-station.m3u?id=1683077
But it doesn't work. Here the log 
```
##CLI.URLSET#: yp.shoutcast.com
##CLI.PATHSET#: /sbin/tunein-station.m3u?id=1683077
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
I (53580) webclient: Webclient socket: 62, errno: 0
I (53580) wifi: Set ps type: 1

##CLI.VOL#: 113
##CLI.URLSET#: 87.118.126.101
##CLI.PATHSET#: /stream
##CLI.PORTSET#: 40143
##CLI.STOPPED# from C_PLIST
I (54920) wifi: Set ps type: 2

I (55000) wifi: Set ps type: 1
```
If i open the link with Firefox for example, I get
```
#EXTM3U
#EXTINF:-1,(#1 - 486/5000) Radio-Carsija
http://87.118.126.101:19406/stream
```
There a trouble for the port with karadio
How can I fix the problem ?

bazooka07, 2020-01-08:
See PR #146 to fix the problem.

karawin, 2020-01-09:
Merci

closed 2020-01-10 by bazooka07