karawin/Ka-Radio32 #145: Ka-Radio32 Url play stream problem

emilio74, 2020-01-04:
Hi. I have installed a ESP32 with this excelent Ka-Radio32.... I try to play this stream " https://icecast.teveo.cu/XjfW7qWN " But i can´t play this radiostation of my country is Fail to connect to the stream. 
what can i do with this trouble? somebody can help?
Thanks a lot :)

bazooka07, 2020-01-04:
Ka-radio32 doesn't work with https protocol.
I'm searching with curl. It seems that is Radio Progreso.
If you continue at radio-browser.info directory
http://www.radio-browser.info/gui/#!/byname/progreso
you can get a stream which works with Ka-Radio :
http://192.99.233.46:8900/;
But I'm not that is the right stream

karawin, 2020-02-10:
Closing. A feature, not a bug ;-)

closed 2020-02-10 by karawin