karawin/Ka-Radio32 #153: Sizeof shoutcast_info > 256 vs getStation()

bazooka07, 2020-01-28:
Sorry but in french is more easy for us.

La liste des stations est un tableau d'enregistrements de 73+116+64+8+16=277 octets chacun (_voir fichier main/include/eeprom.h_).
la fonction getStation() ne lit qu'un bloc de 256 octets (_voir fichier main/include/eeprom.c_).
Cela ne me semble pas cohérent. Si une station a un "domain", un "file" et un "name" au maximum de leurs tailles, il y a certainement un risque de corruption de données.
Il serait peut-être préférable de réduire la taille possible de "domain" et "file"  pour être sûr que la taille d'un enregistrement soit inférieur à 256 octets.

Merci de m'éclairer sur ce point, je ne suis pas un expert en esp32.

karawin, 2020-01-31:
int8_t ovol; // offset volume
	uint16_t port;	//port

Soit 1+2 et non 8+16

Donc 256

bazooka07, 2020-01-31:
Oups, effectivement je mélange les bits et les bytes.
Donc 73+116+64+1+2=256 le compte est bon.

Désolé pour le dérangement.

closed 2020-02-01 by bazooka07

karawin, 2020-02-02:
Pas de pb, avec la somme de code ecrit, une bétise est vite arrivée. Autant vérifier, merci