karawin/Ka-Radio32 #156: two encoders - help addons

ZigUSSS, 2020-02-02:
Hello. Despite numerous attempts with the launch of 2 rotary encoders - no success.
I am asking for a bin file with two encoders (or adding such a file to karadio-addons). First to volume, route to station and display change / stop / play.
I am currently using the standard configuration: ESP32 wroom, 1x rotary encoder, VS1053B, oled 2.42 SSD1309 i2c.
Free pins: 15, 0, 1, 21, 22, 35, 25, 26, 27,
I am very happy with the radio :) I just want to be able to find a more interesting program or program faster
I hope that someone will help a person in poverty.
I apologize for the text translated by the translator

bazooka07, 2020-02-03:
To avoid special binary file, yo can edit one .csv file in the "boards" folder.
Setup for your two encoders by replacing 255 by your io ports :
```
P_BTN0_A,data,u8,255
P_BTN0_B,data,u8,255
P_BTN0_C,data,u8,255
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
```
build the binary image for hardware partition by :
```
sh nvs_partition_generator.sh your-board.csv
```
flash your binary image from build folder to you Esp32 after connecting via USB  port :
```
esptool --chip esp32 write_flash 0x3a2000 build/${your-board}.bin
```

karawin, 2020-02-05:
you need to read the HardwareConfig.md file and configure P_ENC0_x and P_ENC1_x to your needs.

closed 2021-03-05 by karawin