karawin/Ka-Radio32 #160: ttgotm not working

edinsof, 2020-02-07:
hi
Im try to build again on my ttgotm but it dont works 
[image]

when reboot the ttgotm this dont reply. but if build using "standard_adb.bin" wifiharadio now works

karawin, 2020-02-07:
Please log the serial.

nano111, 2020-02-07:
[ttgo_tm.zip](https://github.com/karawin/Ka-Radio32/files/4172918/ttgo_tm.zip)
You can try with these csv files that I made for my ttgo_tm,the one with E in the name is for the ttgo tm with epaper display.
Which version of ttgotm do you have?

edinsof, 2020-02-08:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3564
load:0x40078000,len:9372
ho 0 tail 12 room 4
load:0x40080400,len:5848
entry 0x400806d8
E (580) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (144) main: starting app_main()
I (144) main: RAM left: 200080
E (144) main: Running partition type 0 subtype 16 (offset 0x00010000)
I (144) main: Partition init done...
Log level is now ESP_LOG_ERROR

edinsof, 2020-02-08:
> 
> 
> [ttgo_tm.zip](https://github.com/karawin/Ka-Radio32/files/4172918/ttgo_tm.zip)
> You can try with these csv files that I made for my ttgo_tm,the one with E in the name is for the ttgo tm with epaper display.
> Which version of ttgotm do you have?

please can share ttgotm.bin file?

edinsof, 2020-02-08:
I try to build using ttgotm.bin file from (https://codeload.github.com/karawin/Ka-Radio32/zip/v1.9.4) and it is working now :)
[image]

karawin, 2020-02-10:
So i close, Good

closed 2020-02-10 by karawin