karawin/Ka-Radio32 #161: Button pad config

Mareckil, 2020-02-10:
hello masters, I am having a problem, I would like to use 6 button keypad on my ka-radio32, I have alread done one radio with encoder and it works fine, Now I can't go trough proper configuration for this keypad. 
I would like use 6 or 5 buttons start/stop and station up/down and volume up/down my board ESP32-devkit V1 display tft_lcd ST7735 and VS1053.
Can someone help me with that?

karawin, 2020-02-10:
Hi, please  read https://github.com/karawin/Ka-Radio32/issues/156

Mareckil, 2020-02-11:
Thanks for advice, I have read this thread and manual, so can you please have a look and check if I am thinking well. 

No encoders
P_ENC0_A,data,u8,255
P_ENC0_B,data,u8,0
P_ENC0_BTN,data,u8,0
P_ENC1_A,data,u8,255
P_ENC1_B,data,u8,0
P_ENC1_BTN,data,u8,0 
P_BTN0_A,data,u8,16  // start/stop
P_BTN0_B,data,u8,17  //volume up
P_BTN0_C,data,u8,5   //volume down
P_BTN1_A,data,u8,27  //configured but not used
P_BTN1_B,data,u8,26  //station up
P_BTN1_C,data,u8,22  //station down

karawin, 2020-02-11:
Yes may be if no conflct with the rest of the csv (or default values).
You can now generate and flash the bin