karawin/Ka-Radio32 #162: TTGO-T-Display resolution

tipoya895, 2020-02-11:
Dear KaraWin, you can add a display with a resolution of 135x240 on the ST7789 SPI controller.
Such a display is used in the TTGO-T-Display module. I made the configuration, the display works, I set it to "LCD_SPI_ST7789V 199 // 240x320", half of the screen is not visible.

hevet, 2020-04-29:
I join the request to add this display.

Kobbe1, 2020-05-01:
I would be interested as well. @tipoya895  - what config did you use, could you share it? The standard_adb.bin allows the  webpage to start but I can't get the display to work, even in "half mode" as you describe.
I think the display connection is different as seen here: https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/README.MD

Best,

-T

hevet, 2020-05-01:
key,type,encoding,value
# free field to define the csv
label_space,namespace,,
L_LABEL,data,string,ttgotm.csv
L_COMMENT,data,string,LilyGO/TTGO-TM-ESP32 board .

# Define the gpio used 
gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,17
P_MOSI,data,u8,19
P_CLK,data,u8,18
P_XCS,data,u8,255
P_RST,data,u8,33
P_XDCS,data,u8,2
P_DREQ,data,u8,14
P_ENC0_A,data,u8,32
P_ENC0_B,data,u8,33
P_ENC0_BTN,data,u8,14
P_ENC1_A,data,u8,255
P_ENC1_B,data,u8,255
P_ENC1_BTN,data,u8,255
P_BTN0_A,data,u8,37
P_BTN0_B,data,u8,38
P_BTN0_C,data,u8,39
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
P_I2C_SCL,data,u8,16
P_I2C_SDA,data,u8,5
P_I2C_RST,data,u8,23
P_LCD_CS,data,u8,5
P_LCD_A0,data,u8,16
P_LCD_RST,data,u8,23
P_IR_SIGNAL,data,u8,34
P_I2S_LRCK,data,u8,25
P_I2S_BCLK,data,u8,26
P_I2S_DATA,data,u8,19
P_LED_GPIO,data,u8,22
P_ADC_KBD,data,u8,255
P_BACKLIGHT,data,u8,255
P_JOY_0,data,u8,255
P_JOY_1,data,u8,255

option_space,namespace,,
O_LCD_TYPE,data,u8,199
O_LCD_ROTA,data,u8,255
O_LCD_OUT,data,u32,0
O_DDMM_FLAG,data,u8,255

custom_ir_space,namespace,,
K_UP,data,string,
K_LEFT,data,string,
K_OK,data,string,
K_RIGHT,data,string,
K_DOWN,data,string,
K_0,data,string,
K_1,data,string,
K_2,data,string,
K_3,data,string,
K_4,data,string,
K_5,data,string,
K_6,data,string,
K_7,data,string,
K_8,data,string,
K_9,data,string,
K_STAR,data,string,
K_DIESE,data,string,
K_INFO,data,string,

Kobbe1, 2020-05-01:
Thank you, that was quick.
Will try it - you don't happen to have the bin file around maybe?

hevet, 2020-05-01:
[ttgotm_moje.zip](https://github.com/karawin/Ka-Radio32/files/4566382/ttgotm_moje.zip)

hevet, 2020-05-01:
[ttgotm_moje.zip](https://github.com/karawin/Ka-Radio32/files/4566391/ttgotm_moje.zip)

Kobbe1, 2020-05-01:
Tried, not working yet, also from 2nd zip.
It seems above list has a typo maybe?

P_RST,data,u8,33
...
P_ENC0_B,data,u8,33

RST should be on 23 I think?

hevet, 2020-05-01:
Sorry, I sent this configuration, it should be correct.

key,type,encoding,value

label_space,namespace,,
L_LABEL,data,string,standard_adb.csv
L_COMMENT,data,string,This is the standard default configuration as in the builtin conf.

gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,2
P_MOSI,data,u8,19
P_CLK,data,u8,18
P_XCS,data,u8,32
P_RST,data,u8,12
P_XDCS,data,u8,33
P_DREQ,data,u8,34
P_ENC0_A,data,u8,14
P_ENC0_B,data,u8,17
P_ENC0_BTN,data,u8,255
P_ENC1_A,data,u8,255
P_ENC1_B,data,u8,17
P_ENC1_BTN,data,u8,5
P_BTN0_A,data,u8,255
P_BTN0_B,data,u8,255
P_BTN0_C,data,u8,255
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
P_I2C_SCL,data,u8,16
P_I2C_SDA,data,u8,5
P_I2C_RST,data,u8,23
P_LCD_CS,data,u8,5
P_LCD_A0,data,u8,16
P_LCD_RST,data,u8,23
P_IR_SIGNAL,data,u8,21
P_I2S_LRCK,data,u8,25
P_I2S_BCLK,data,u8,26
P_I2S_DATA,data,u8,22
P_TOUCH_CS,data,u8,255
P_JOY_0,data,u8,255
P_JOY_1,data,u8,255
P_LED_GPIO,data,u8,255
P_ADC_KBD,data,u8,255
P_BACKLIGHT,data,u8,4

option_space,namespace,,
O_LCD_TYPE,data,u8,199
O_LCD_ROTA,data,u8,0
O_LCD_OUT,data,u32,0
O_DDMM_FLAG,data,u8,0

custom_ir_space,namespace,,
K_UP,data,string,
K_LEFT,data,string,
K_OK,data,string,
K_RIGHT,data,string,
K_DOWN,data,string,
K_0,data,string,
K_1,data,string,
K_2,data,string,
K_3,data,string,
K_4,data,string,
K_5,data,string,
K_6,data,string,
K_7,data,string,
K_8,data,string,
K_9,data,string,
K_STAR,data,string,
K_DIESE,data,string,
K_INFO,data,string,

[ttgo.zip](https://github.com/karawin/Ka-Radio32/files/4566458/ttgo.zip)

Kobbe1, 2020-05-01:
Yes!!
Thanks a lot.

Condor-XYZ, 2021-01-03:
Hello hevet
I also have the TTGo T-Display module and tried the TTGO.bin. however, half of the picture is missing because the resolution is not correct. only 135x240 pixels are required and not 240x320 as stored in driver 199.
What am I doing wrong ?
[image]

hevet, 2021-01-03:
You're not doing anything wrong.  If you read the posts carefully, this is how the predecessor works.  Unfortunately, there is no support for this resolution.

LukyanovVE, 2021-01-15:
I join the request to add this display. Thank you

mfsysprog, 2021-02-09:
I got this to work changing the 240x320 files that go with the configured type 199. The t-display board uses a SP7789V controller, which is for a 240x320 device, but the lcd is 135x240.
see:
https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/a4d5c5c0eb28b4421dcaf068ce65d8c392a91264/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h
The CGRAM_OFFSET that is mentioned in the above link, translates to a 40, 53 offset in landscape mode.
https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/a4d5c5c0eb28b4421dcaf068ce65d8c392a91264/TFT_eSPI/TFT_Drivers/ST7789_Rotation.h

I've attached the diff output for the changes I made. It also includes the resulting bin for the hardware config and the generated KaRadio32.bin. The display part for this image will work, but I have not looked at any of the other configuration questions that were asked, so I'm not sure if it is fully functional otherwise. I've also inverted the colors, I personally like that better.

Reason I did not provide a pull request is that in ```main/addonucg.c``` the ```case text``` logic will select a font based on the height of 240 that is too big for this display. So as a quick fix I changed the 240 selection to the default ```ucg_font_6x13_mf``` font that seems nice for this display, but perhaps the are some bigger fonts that would also still fit.
For a permanent fix perhaps the inY should be taken into account also.

The git diff output, the hardware config csv and resulting bin and the KaRadio32.bin:
[tdisplay.zip](https://github.com/karawin/Ka-Radio32/files/5953843/tdisplay.zip)

To give an idea how the end result looks:
[image]

@karawin ; perhaps this gives enough info to come up with a permanent solution, if you need additional info feel free to contact me.

karawin, 2021-02-09:
Hi,

Thanks, great job.
I will integrate it by creating a new LCD_SPI type may be _ST7789S
And yes, i will add the Y value much specific in this case.

Well done

karawin, 2021-02-09:
I made a prerelease with 
#define LCD_SPI_ST7789T			204 // 135x240

mfsysprog, 2021-02-10:
I originally took the same approach:
```
       case LCD_SPI_ST7789T:
                ucg_Init(&ucg, ucg_dev_st7789_18x240x320, ucg_ext_st7789_18, ucg_com_hal);
                break;
```
But I couldn't get it to fit onto the screen. It was only later that I realized that it was the font size that forced half of the information of the screen. By then I had changed so much that I fell back to a clean clone and just changed the type 199 config.
Some observations that I made along the way that might be helpful:
- I just realized that in this case it seems to me that the offset should be in the ucg_dev_tft* file and not in the ucg_dev_ic*. I'm guessing that for alternative sized screens (compared to the 'native' resolution of the SP7789V controller) just not all source and gate pins are connected and the offset 'places' the screen contents into the correct positions in the buffer for the pins that are connected to the tft.
- My initial guess was that I had to change the offset in the ucg_st7789_set_pos_dir*_seq[] arrays. I'm guessing these are for positioning stuff in (inversed) portrait and landscape mode? I had a hard time understanding what the UCG_VARX and UCG_VARY macro's are doing. I only later realized that the 0x0a0 and 0x0b0 that they use are not actual bytes send to the device, but are used for the switch/case in ```ucg_com_SendCmdSeq``` in ```ucg_com_msg_api.c```. The ```s, a, o``` in the define I'm guessing are shift, address and offset? 
-  I frequently see UCG_VARX(0,0x0ff, 0) and UCG_VARY(0,0x0ff, 0). Looking in the datasheet is says that all bytes send outside of the buffer are ignored, so I'm guessing this should mean 'to the end of the buffer', but I don't understand how this works when the screen is larger than 0xff in the X or Y direction. But perhaps my understanding of it is wrong.

karawin, 2021-02-15:
Done but not tested
Thanks

closed 2021-02-15 by karawin