karawin/Ka-Radio32 #163: Jamming on the new version 1.9R7

hevet, 2020-02-13:
I have updated two radios to the latest version, however after the update the playback stutters or stops for a moment and restarts. The problems are on two radios that I made. It doesn't even help to load all files clean. Return to version 1.9R6 solves the problems. Maybe you need to upload something differently, or someone also has the same problem?

yo2ldk, 2020-02-13:
the same on my modules

karawin, 2020-02-14:
Yes, sorry, a mistake for configuration with VS153. Please update again.

hevet, 2020-02-14:
Thanks to dragging, but there is another problem, now it is much slower on the display, I use ST7735, resolution 240x320.

karawin, 2020-02-14:
Yes, i noticed it. It seems that we are close to limit of the cpu. I am working to found a solution for all configurations.
The worst is DAC with AAC decoding and a 240x320 color display.

hevet, 2020-02-14:
Thanks for the work you do, I have a second radio with ST7735 and a resolution of 128x160, and here it is definitely better. I hope you can solve the problem, for now I will stay with version 1.9R6 on the 240x320 display.

karawin, 2020-02-14:
Now at full speed. Update again.
Thanks for your feedback.

hevet, 2020-02-14:
Thank you very much for the quick resolution of problems, it now works as in the previous version.

closed 2020-02-15 by karawin