karawin/Ka-Radio32 #166: Mute signal on GPIO, wifi signal strength and db scale.

krzsiek, 2020-02-19:
Hi, thanks for your project, is awesome.
I have an 3 ideas. Can you give a logic Mute output to GPIO? I use the built-in DAC, sometimes it is enough, but when switching stations, there are cracks. Many amplifiers IC have a mute input, where  on hight signal it mutes the amp output. If mute signal lasting from the end of decoding the previous station to the moment when the next plays it would be a nice improvement.

Second idea, a volume scale in dB instead of 1-254 on lcd. Like in web gui. Similarly, instead of 254 steps, for example 25 would be easier to use.  3dB steps control, from -80, -77, -76 .... to 0 dB.

Third idea. Info about the strength of the wifi signal on the screen next to the IP number. I mean the clock screen and the screen when changing stations. Something like simulation on photo below.

Regards!

[image]

pilnikov, 2020-02-20:
++++

nadim24, 2020-02-20:
++++

karawin, 2020-02-20:
Thanks for your ideas.
Only the 3 will be done.

Third Point:Will be something like that:
[image]

krzsiek, 2020-02-21:
Thanks. 
Will the battery indicator be configurable using commands to turn off? When you use the main power source, it is unnecessary. It also seems, that as the bottom line is only for the IP address, it looks clearer as it is in the middle as before.