karawin/Ka-Radio32 #167: vs1053 noise (solution)

rolandinchania, 2020-02-20:
Browsing the internet I found a solution to remove noise  and ground loop audio problems when useing the vs1053 dac board with an external audio amplifier. 

https://electronics.stackexchange.com/a/186921

important: the wireing diagram at the and of the article. Don't use the GBUF signal as ground with an external amp because its for the headphone output. Use  GND instead.

Big success for my project. !!!

karawin, 2020-02-22:
Yes or simply a couple of isolation transformers.

rolandinchania, 2020-02-22:
Thanks. I made 2 simple changes to my vs1053 board. 
1. cut the gbuf line
2. add new groundwire (black)

[image]

cheise, 2020-02-22:
I would have the same problem, thanks for this solution. 

cu

rolandinchania, 2020-02-29:
Thank you, it's cool !!!

rolandinchania, 2020-02-29:
Thank you all.

closed 2020-02-29 by rolandinchania

Sera26, 2021-03-30:
Hello,

I really like your solution and want to perform it.
But I have the red version of VS1053 board.
Could you help me which lines need to be cutted and bridged?
Thank you in advance.

[image]

rolandinchania, 2021-03-30:
Hi,

I don't know the board but similar to the green board I suggest:

Cut where the little black line is. then connect from the connector to ground.

for example ground is under the green dot when you remove the paint on the board.

Regards Roland

 

ps. I enclosed a picture

Sera26, 2021-03-30:
Hi,

Thanks for quick response.
If I well understand...need to cut line of the current gnd of speaker jack and bridged to main GND of the board (your dots)?

rolandinchania, 2021-03-30:
exactly 😊-- Sent from my Android phone with WEB.DE Mail. Please excuse my brevity.

Sera26, 2021-03-30:
No problem :) Thank you for support :)

cheise, 2021-03-31:
[image]

;)

Sera26, 2021-03-31:
Yes, it is, thank you very much :)