karawin/Ka-Radio32 #169: Screensaver to prevent OLED burnout.

evilslon, 2020-02-26:
Hello! 
I use a monochrome OLED display in my KaRadio32, but there's one problem - the display burns out when the device is in sleep mode (when it shows the clock). Could you make something similar to a screensaver (for example, for clock numbers changing the position on the screen) to prevent pixels from burning out? That would be great!

Thank you for your excellent work!

[image]
[image]

hjgode, 2021-01-07:
I also need some help to make oled display either dim or mover around, when stopped playing. Some kind of screen off or screen saver would be good.

hjgode, 2021-01-08:
In the meantime I will use the command 
          sys.lcdstop("5") 
to shutdown the OLED after 5 seconds when I stopped playing. There is also an option to make the display go off in play/normal mode: 
          sys.lcdout("10") 
will switch off the display in play mode after 10 seconds of being idle.

Oled displays support a contrast settings, that may be usefull to 'dim' the display after some idle time. But this would require some extra coding...

closed 2021-02-22 by karawin