karawin/Ka-Radio32 #172: Can't build the latest release (1.9R7) with esp-idf v3.3

elabree, 2020-04-14:
Hi,

I started recently to discover the possibliities of the EPS32.

But getting karadio working seems a bit difficult. I followed the steps in README.md and HardwareConfig.md. Result of make flash:

"C:/msys32/home/Egon/esp/Ka-Radio32/main/app_main.c:660:22: error: invalid initializer
   ip_addr_t ipdns0 = dns_getserver(0);
                      ^
make[1]: *** [/home/Egon/esp/esp-idf/make/component_wrapper.mk:290: app_main.o] Fout 1
make: *** [C:/msys32/home/Egon/esp/esp-idf/make/project.mk:552: component-main-build] Fout 2
"
I am not sure where to look. My mingw32 with esp-idf seems to be OK since I can built other projects correctly.

Any advise ?
Kind Regards,
Egon :-)

karawin, 2020-04-16:
At C:\msys32\home\jp\esp\esp-idf\components\lwip\lwip\src\include\lwip\dns.h
Line 109
ip_addr_t       dns_getserver(u8_t numdns);

May be a compiler issue?

Try 
ip_addr_t ipdns0 ;
ipdns0= dns_getserver(0);

In place of 
ip_addr_t ipdns0 = dns_getserver(0);

elabree, 2020-04-17:
Hi,
Thank you for the advise.

Egon@Q910 MINGW32 ~/esp/esp-idf
$ git describe
v3.3.2-26-g1ba3bf297
Egon@Q910 MINGW32 ~/esp/esp-idf
$ git status -uno
On branch release/v3.3
Your branch is up to date with 'origin/release/v3.3'.

The result after making the suggested change:
"$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/Egon/esp/esp-idf/requirements.txt are satisfied.

App "KaRadio32" version: 1.9.7
CC build/main/app_main.o
C:/msys32/home/Egon/esp/Ka-Radio32/main/app_main.c: In function 'start_network':
C:/msys32/home/Egon/esp/Ka-Radio32/main/app_main.c:661:7: error: 'ipdns0' undeclared (first use in this function)
       ipdns0= dns_getserver(0);
       ^
C:/msys32/home/Egon/esp/Ka-Radio32/main/app_main.c:661:7: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [/home/Egon/esp/esp-idf/make/component_wrapper.mk:290: app_main.o] Fout 1
make: *** [C:/msys32/home/Egon/esp/esp-idf/make/project.mk:552: component-main-build] Fout 2
"
Tried to use an old version of esp-idf v3.3:
git checkout 12d639e5b204671a832ba577b9974ecb276e6380
git submodule update
git submodule init
-> That seems to help: make flash builds the project !

So you are right: the latest esp-idf seems to have an issue.
Thanks,
Egon :-)

r3d4, 2020-05-10:
ip_addr_t *ipdns0 = dns_getserver(0);
printf("\nDNS: %s  \n",ip4addr_ntoa(( struct ip4_addr* ) ipdns0));

karawin, 2020-05-10:
right, thanks

karawin, 2020-05-10:
In $ git describe
v3.3.2-107-g722043f73
At C:\msys32\home\jp\esp\esp-idf\components\lwip\lwip\src\include\lwip\dns.h
Line 109
ip_addr_t dns_getserver(u8_t numdns);
Changed to
const ip_addr_t* dns_getserver(u8_t numdns);

So now you are right. The new code is
		ip_addr_t *ipdns0 = dns_getserver(0);
//		ip_addr_t ipdns1 = dns_getserver(1);
		printf("\nDNS: %s  \n",ip4addr_ntoa(( struct ip4_addr* ) ipdns0));

Thanks