karawin/Ka-Radio32 #175: SPDIF output

Vasilis-7, 2020-05-03:
I am thinking the possibility to add it on my home amplifier set. 
Is it possible, to add an SPDIF output ?
In order to have the best audio result i think the digital way is better than the analog.

r3d4, 2020-05-04:
It would be great to have SPDIF output. Maybe something like this Software-SPDIF would be possible:
https://github.com/earlephilhower/ESP8266Audio/blob/master/src/AudioOutputSPDIF.cpp

This would allow high quality playback with almost no additional hardware needed.

r3d4, 2020-05-10:
I ported the SPDIF algorithm from ESP8266Audio to Ka-Radio32 and created a pull request: https://github.com/karawin/Ka-Radio32/pull/176
It only needs an optical Transmitter at the I2S data pin (or even a LED + resistor might work). Tested it with 44.1 an 48KHz for 2 days, works quite well.

r3d4, 2020-05-14:
I reworked (deleted) SPDIF branch an created new pull request: https://github.com/karawin/Ka-Radio32/pull/178

karawin, 2020-05-15:
I tried to merge, but conflict with the working change. I had to revert. Please work on it now. Sorry.

marcinmaster999, 2020-05-16:
Thx 

> I reworked (deleted) SPDIF branch an created new pull request: #178

Thx r3d4.This patch works great for 1.9.7.

r3d4, 2020-05-18:
@karawin 
Please ignore pull request:  #176 this branch was delete by me. 
All necessary changes are in #178. It should work to merge #178 direct to actual master.
I don't get conflicts here.

Vasilis-7, 2020-08-17:
Hi. Is it working the SPDIF out?

karawin, 2021-01-29:
Done

closed 2021-01-29 by karawin