karawin/Ka-Radio32 #193: https stream support

milenmalinov, 2020-11-25:
Hello, is it possible to add ssl support. Many of the stations are already on https://

hevet, 2020-11-28:
Four topics below https://github.com/karawin/Ka-Radio32/issues/184

milenmalinov, 2020-11-29:
> Four topics below #184

Yes,
opened on Jul 17 by hevet and no reasonable answer.

hevet, 2020-11-29:
Nothing meaningful has been written because there is no such possibility yet.

karawin, 2020-12-29:
right ;-)

pilnikov, 2020-12-29:
https://serverdoma.ru/viewtopic.php?f=70&t=1178&p=9143#p9129
in translation (in short) - possible

karawin, 2021-01-01:
This will may be feasible on a wrover only because one htmls  connexion needs 40k of heap.

karawin, 2021-01-29:
Done ;-)

closed 2021-01-29 by karawin

hevet, 2021-01-29:
Will this stream work on this version? https://online.nasze.fm:9443/stream

milenmalinov, 2021-01-29:
> Will this stream work on this version? https://online.nasze.fm:9443/stream

No! The stream is mpeg and will not work.

karawin, 2021-01-29:
No, it say
-112 mp_exptmod error state

Don't know what it means.