karawin/Ka-Radio32 #194: Working configuration for ESPDUINO-32 and VS1053 MP3 Shield

Zdenalus, 2020-12-06:
Hi, I can't get it to work, still:

##CLI.ICY4#: Invalid audio output. VS1053 not found
E (7341) main: Invalid audio output. VS1053 not found

I modified standard_minimal to:

gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,19
P_MOSI,data,u8,23
P_CLK,data,u8,18
P_XCS,data,u8,27
P_RST,data,u8,12
P_XDCS,data,u8,14
P_DREQ,data,u8,34 

[image]
[image]

Thank you
Zdenek

hevet, 2020-12-10:
Change P_DREQ,data,u8,34 on P_DREQ,data,u8,26. 
I have the same configuration

Zdenalus, 2020-12-11:
I have changed it, but still no success

Wiring:
[image]

Here changes to standard_minimal.csv:
[image]

hevet, 2020-12-11:
According to the DREQ drawing, you connect to the RX or TX pin, and it must be to one further, the one with "2."
[image]

Zdenalus, 2020-12-11:
Ok, I'll try it. CSV configuration is correct? DREQ to 26? But connect to tx or tx on espduino?

hevet, 2020-12-11:
CSV configuration is correct.  You don't need to connect RX and TX.

Zdenalus, 2020-12-12:
So I connected DREQ (2) with ESPDUINO 26, What should I connect to TX or RX?
Still not working... ;(

hevet, 2020-12-12:
I wrote that you do not connect anything.  For me, this configuration works on these settings, so I can't help you with anything else.

hevet, 2020-12-12:
Użyj jeszcze takiej konfiguracji

key,type,encoding,value

label_space,namespace,,
L_LABEL,data,string,standard_adb.csv
L_COMMENT,data,string,This is the standard default configuration as in the builtin conf.

gpio_space,namespace,,
K_SPI,data,u8,2
P_MISO,data,u8,19
P_MOSI,data,u8,23
P_CLK,data,u8,18
P_XCS,data,u8,27
P_RST,data,u8,12
P_XDCS,data,u8,14
P_DREQ,data,u8,26
P_ENC0_A,data,u8,25
P_ENC0_B,data,u8,17
P_ENC0_BTN,data,u8,16
P_ENC1_A,data,u8,32
P_ENC1_B,data,u8,33
P_ENC1_BTN,data,u8,15
P_BTN0_A,data,u8,255
P_BTN0_B,data,u8,255
P_BTN0_C,data,u8,255
P_BTN1_A,data,u8,255
P_BTN1_B,data,u8,255
P_BTN1_C,data,u8,255
P_I2C_SCL,data,u8,22
P_I2C_SDA,data,u8,21
P_I2C_RST,data,u8,2
P_LCD_CS,data,u8,21
P_LCD_A0,data,u8,22
P_LCD_RST,data,u8,2
P_IR_SIGNAL,data,u8,255
P_I2S_LRCK,data,u8,255
P_I2S_BCLK,data,u8,255
P_I2S_DATA,data,u8,255
P_TOUCH_CS,data,u8,255
P_JOY_0,data,u8,255
P_JOY_1,data,u8,255
P_LED_GPIO,data,u8,4
P_ADC_KBD,data,u8,255
P_BACKLIGHT,data,u8,255

option_space,namespace,,
O_LCD_TYPE,data,u8,192
O_LCD_ROTA,data,u8,0
O_LCD_OUT,data,u32,0
O_DDMM_FLAG,data,u8,1

custom_ir_space,namespace,,
K_UP,data,string,
K_LEFT,data,string,
K_OK,data,string,
K_RIGHT,data,string,
K_DOWN,data,string,
K_0,data,string,
K_1,data,string,
K_2,data,string,
K_3,data,string,
K_4,data,string,
K_5,data,string,
K_6,data,string,
K_7,data,string,
K_8,data,string,
K_9,data,string,
K_STAR,data,string,
K_DIESE,data,string,
K_INFO,data,string,

Zdenalus, 2020-12-14:
It's working now! Thank you very much!

closed 2020-12-14 by Zdenalus