karawin/Ka-Radio32 #196: ESP32 Audio Kit Support?

briankdlee, 2020-12-26:
I downloaded Ka-Radio32 to ESP32 Audio Kit V2.2.
(You can buy it from aliexpress,
https://www.aliexpress.com/wholesale?catId=0&SearchText=esp32-audio-kit)

I can connect the web interface of Ka-Radio32.
But, when I pressed play button on the web interface, I got the following error(console output).

--------------------------------------------------------------------------------
##CLI.META#: 
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40090fce  PS      : 0x00060333  A0      : 0x80093db5  A1      : 0x3ffe30e0  
A2      : 0xff040000  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060320  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffe30e0  
A10     : 0x00000003  A11     : 0x00060323  A12     : 0x00060320  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xff040000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: b9bed1b283796ed068372b46f977cf9989867507d2615e3c59a83f572fae1874

Backtrace: 0x40090fce:0x3ffe30e0 0x40093db2:0x3ffe3110 0x400881b9:0x3ffe3130 0x4008820b:0x3ffe3150 0x40088d95:0x3ffe3170 0x4000beaf:0x3ffe3190 0x40171c1b:0x3ffe31b0 0x40092599:0x3ffe31d0

--------------------------------------------------------------------------------

Is there anyone using this board, ESP32 Audio Kit?

Thanks in advance.

pilnikov, 2020-12-26:
i used
Welcome to facebook in group Karadio
https://www.facebook.com/groups/162949914181385/?_rdc=1&_rdr
[image]

briankdlee, 2020-12-27:
This is exact answer what I expected. Karadio facebook group will be helpful for me. Thank you for the comment.

closed 2021-01-06 by briankdlee