karawin/Ka-Radio32 #198: Rotary Encoder doesn't change stations

clparrish, 2021-01-11:
Radio works great. I do not have a display. I have one rotary encoder implemented, which works as follows:
1. Rotating changes volume.
2. Push mutes output as long as it is held.
3. Push and rotate changes volume. This is manifested when released.
I thought it was supposed to work like this:
1. Rotating changes volume.
2. Push mutes output until it is pushed again.
3. Push and rotate selects another station. This is manifested when released.

karawin, 2021-01-29:
Something is wrong in your configuration.
2/ is abnormal
3/ is abnormal

5/ is a start/stop
3 and 6 ???

clparrish, 2021-01-29:
Thanks for your comments. Configuration is whatever is present in the upload, I don't know how to change it. I've reloaded several times; the behavior is the same each time the firmware is uploaded. Still mystified.

karawin, 2021-01-30:
Without display, two encoders may be a solution, but strange.

clparrish, 2021-03-04:
I found that there was a wiring error. The rotary control works correctly now. Thanks to karawin for the help.

closed 2021-03-04 by clparrish