karawin/Ka-Radio32 #200: wolfSSL_connect - error

fortify74, 2021-02-03:
Hi KaraWin, a very good job!
this radio station https://live.westcityradio.ro:8443/aac does not work.

LOG:
I (305530) webclient: socket: 58
I (305540) webclient: ip: 729cd6bc   ADDR:188.214.156.114
E (306400) webclient: Failed to connect to wolfSSL
E (306400) webclient: wolfSSL_connect err: -112
E (306500) webclient: SSL Cleanup 1 Client socket: 58_

Thanks

karawin, 2021-02-04:
Yes, some url cannot  connect.
Unfortunately, i don't know why.
A problem of cryptothings ?
Sorry and thanks.

mfsysprog, 2021-02-11:
This site uses a certificate with a 4096 bit sha key. So it is the same error as in https://www.wolfssl.com/forums/topic1109-solved-rsafunction-mpexptmode-memoryconfig-problem-error.html
I could get it to load using a hardcoded ```#define FP_MAX_BITS 8192``` in ```components/wolfssl/wolfssl/wolfcrypt/settings.h``` and increasing the stack size in ```main/app_main.c``` to ``` xTaskCreatePinnedToCore(clientTask, "clientTask", 3400, NULL, PRIO_CLIENT, &pxCreatedTask,CPU_CLIENT); ``` 
Without that I got a stack overflow, the value 3400 was just arbitrarily chosen to be bigger than the original 3100.
From the serial console I can see that the ssl error is gone. I don't have the hardware yet to actually play the aac stream, but I think it works.
I added a log callback function as per https://www.wolfssl.com/docs/wolfssl-manual/ch8/ to get the wolfssl messages on the serial console. I just disregarded the loglevel, a correct implementation would look at that to decide ESP_LOGx level.
```
--- a/main/webclient.c
+++ b/main/webclient.c
@@ -1276,6 +1276,10 @@ uint8_t bufrec[RECEIVE+20];
     /* declare wolfSSL objects */
     WOLFSSL_CTX *ctx;
     WOLFSSL *ssl;
+
+void wolfSSL_log_function(const int logLevel, const char *const logMessage){
+    ESP_LOGD(TAG,"WOLFSSL:%s\n",logMessage);
+}
        
 void clientTask(void *pvParams) {
        portBASE_TYPE uxHighWaterMark;
@@ -1299,6 +1303,7 @@ void clientTask(void *pvParams) {
        if (wolfSSL_Init() != WOLFSSL_SUCCESS) {
                ESP_LOGE(TAG,"ERROR: failed to init WOLFSSL\n");}
        wolfSSL_Debugging_ON();
+        wolfSSL_SetLoggingCb(wolfSSL_log_function);
        /* Create and initialize WOLFSSL_CTX */
        if ((ctx = wolfSSL_CTX_new(wolfSSLv23_client_method())) == NULL) {
                ESP_LOGE(TAG,"ERROR: failed to create WOLFSSL_CTX\n");
```

karawin, 2021-02-11:
Great job, it works. 
Soon in the 2.1 R0 ;-)

Thanks

karawin, 2021-02-12:
Yes the debugging log is a little verbose ;-)
I will may be add a command  wolfssl debugging on and off

fortify74, 2021-02-12:
Thanks a lot, I look forward to version 2.1 !

closed 2021-02-15 by fortify74