karawin/Ka-Radio32 #202: V2.0 R1 error

Jambola77, 2021-02-11:
I have upgraded to version V2.0 R1 and now the playing stopps directly after it started:

Jambola77, 2021-02-11:
```
##CLI.NAMESET#: 7 Antenne Vorarlberg Lounge
##CLI.URLSET#: http://webradio.antennevorarlberg.at
##CLI.PATHSET#: /lounge.m3u
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 220
##CLI.URLSET#: http://HTTP
##CLI.PATHSET#: /1.0 200 OK
##CLI.PORTSET#: 80
##CLI.STOPPED# from C_PLIST
##CLI.STOPPED# from clientConnect
##CLI.ICY0#: gethostbynam
```
It worked before and the station URL is correct. Any idea?

karawin, 2021-02-11:
For non https station the url must be webradio.antennevorarlberg.at without http://
WRONG
##CLI.URLSET#: http://webradio.antennevorarlberg.at

OK
##CLI.URLSET#: webradio.antennevorarlberg.at
##CLI.PATHSET#: /
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 56
##CLI.PLAYING#

Jambola77, 2021-02-11:
[image]

I cannot add a station without http in the list. When I remove the http://, I have a blank URL.

karawin, 2021-02-11:
It is normal to have the http:// in the full url.
It is parsed correctly in Url, Port Path.
You can  save it as it is.

Jambola77, 2021-02-11:
No, it does not work. Still the same error.

Jambola77, 2021-02-11:
I have now inserted "webradio.antennevorarlberg.at/lounge.m3u" in the radio tab in the field "Instant Play" and got the same error (gethostbyname).

karawin, 2021-02-11:
yes that strange. I will try to find. Thanks

Jambola77, 2021-02-11:
It worked perfectly with V1.9 R9.

karawin, 2021-02-11:
Found. Update. It is a 2.1 R0 but not official yet ;-)
Thanks

Jambola77, 2021-02-15:
Issue ist fixed. Thanks a lot!

closed 2021-02-15 by Jambola77