karawin/Ka-Radio32 #204: Bad display of characters

Janda008, 2021-02-14:
Hello. There is no solution to this problem. Characters are displayed incorrectly. Thanks for the info.
I'm sorry for my bad english.
[image]

karawin, 2021-02-15:
Name you gave to the station and station url please?

Janda008, 2021-02-15:
Station Name "Blaník" and url "https://playerservices.streamtheworld.com:443/api/livestream-redirect/RADIO_BLANIKCZ_128.mp3"

It doesn't matter what station is playing, if these characters are to be displayed in the text "ĚŠČŘŽŤŮ / ěščřžťů" they will be displayed incorrectly on color lcd. There is no problem with monochrome LCDs. It is possible to simulate that these characters are called radio stations. Thank you for your help.

karawin, 2021-02-15:
I made a basic mapping in the prerelease.
The problem is that the font doesn't include the full utf8 latin char . the B/w driver contains it.
So I made a rough mapping (in progress)
TEst it if you want.
command:
sys.prerelease

Janda008, 2021-02-16:
I just tested the character editing and found that most of it is fine. 
Summary: 
Ě = E  ě = e  - OK
Š = S  š = s   - OK
Č = C  č = c  - OK
Ř = R  ř = r   - OK
Ž = Z  ž = z   - OK
Ů = U ů = u  - OK
Ť = d   ť =  e - NG

Only the "Ť / ť " is angry otherwise it's great. Thanks for a good job!

karawin, 2021-02-18:
DOne in 2.1R1.
Far from perfect, but I will improve it  time to time.

Janda008, 2021-02-19:
[image]

Thank you once again!

closed 2021-02-19 by Janda008

kauczu, 2023-12-30:
> I made a basic mapping in the prerelease. The problem is that the font doesn't include the full utf8 latin char . the B/w driver contains it. So I made a rough mapping (in progress) TEst it if you want. command: sys.prerelease

Hi there! 
Any chance to update mapping utf8 localized chars? There is a problem to display Polish characters, such as: Ś ś Ą ą Ż ż Ź ź Ć ć Ń ń Ę ę Ó ó Ł ł
Or if you out of time or something just point me to where to find this piece of code and I will try my best ;)