karawin/Ka-Radio32 #206: playback problems for Bitrate radio stations: 320kB/s

fortify74, 2021-02-16:
The following stations do not work correctly with the latest version (2.0) both are 320 kB/s:
  http://streaming.hotmix-radio.net:80/hotmixradio-hits-64.mp3
  http://158.69.114.190:8014/stream
  
**LOG version 2.0 - **buffer fill is under 20%****
_##CLI.ICY2#: SHOUTcast DNAS\/posix(linux x64) v2.5.5.733<BR>
##CLI.ICY3#: http:\/\/www.magicfmonlineradio.com
##CLI.ICY4#: 80s, Dance
##CLI.ICY5#: 320
D (32252) webclient: t1: 0x3ffc772b, cstatus: 2, icyfound: 1  metad:8192 Metaint:8192

D (32252) webclient: newlen: 0   len: 543   chunked:0  pdata:3ffc7510
##CLI.PLAYING#
D (32462) renderer: Renderer volume:  4800
D (32502) webclient: clientReceiveCallback: pdata: 3ffc7510, pdataend: 3ffc7d10, len: 2048
D (32502) webclient: mt len:2048, clen:2048, metad:0 ,&l:3ffc7510, l:48, inpdata:3ffc7510, rest:1999, str: StreamTitle='Pretenders - Don't Get Me Wrong';
##CLI.META#: Pretenders - Don't Get Me Wrong
D (33482) webclient: clientReceiveCallback: pdata: 3ffc7510, pdataend: 3ffc7d10, len: 2048
D (33482) webclient: mt len:2048, clen:2048, metad:145 ,&l:3ffc75a1, l:48, inpdata:3ffc7510, rest:1854, str: StreamTitle='Abc - The Look Of Love';
##CLI.META#: Abc - The Look Of Love
D (37902) webserver: watermark serverClientTask: 144  324
I (48412) audio_player: Buffer fill 10%, 5531 // 51200 bytes
I (59572) audio_player: Buffer fill 7%, 4095 // 51200 bytes
I (69612) audio_player: Buffer fill 9%, 4942 // 51200 bytes
I (81082) audio_player: Buffer fill 34%, 17819 // 51200 bytes
I (92282) audio_player: Buffer fill 17%, 9027 // 51200 bytes_

With version ./binaries/KaRadio32_19R7.bin it works very well.

**LOG version 1.9 R7 - **buffer fill > 80%****
_##CLI.ICY2#: SHOUTcast DNAS\/posix(linux x64) v2.5.5.733<BR>
##CLI.ICY3#: http:\/\/www.magicfmonlineradio.com
##CLI.ICY4#: 80s, Dance
##CLI.ICY5#: 320
D (12851) webclient: t1: 0x3ffc781b, cstatus: 2, icyfound: 1  metad:8192 Metaint:8192

D (12851) webclient: newlen: 0   len: 543   chunked:0  pdata:3ffc7600
D (13071) renderer: Renderer volume:  4800
##CLI.PLAYING#
D (13171) webclient: clientReceiveCallback: pdata: 3ffc7600, pdataend: 3ffc7b9c, len: 1436
D (13171) webclient: mt len:1436, clen:1436, metad:1012 ,&l:3ffc79f4, l:48, inpdata:3ffc7600, rest:375, str: StreamTitle='Commodores - Nightshift';
##CLI.META#: Commodores - Nightshift
D (13331) webserver: Heap size: 44388
D (13341) webserver: websocketAccept socket: 56
D (13341) webserver: watermark serverClientTask: 49c  1180
D (13401) audio_player: RAM left 44740
D (13401) audio_player: decoder task created: vsTask
I (20061) audio_player: Buffer fill 84%, 43286 // 51200 bytes
I (27681) audio_player: Buffer fill 99%, 51144 // 51200 bytes
I (35521) audio_player: Buffer fill 99%, 50733 // 51200 bytes_

karawin, 2021-02-17:
Working on it.
Thanks

karawin, 2021-02-18:
Done in 2.1R1

Thanks

fortify74, 2021-02-20:
works 
Thanks

closed 2021-02-20 by fortify74