karawin/Ka-Radio32 #210: 16x2 Scroll Speed Change

usermoko, 2021-03-02:
Hi, i am using 16x2 lcd i2c, but the scroll speed is very fast and display does not react very well. how can i decrease it. the display i am using is built in dab radio philips ae5020. it is cog(chip on glass) type. Changed the library from LiquidCristal_I2C.h to ST7032.h in karadio addon works fine. The only issue is scroll speed.

karawin, 2021-03-03:
You are not in the good project, but i am pretty sure that you can find the answer yourself
    if (loopcount++ == 0x3fff)
    {
	    loopcount = 0;
		scroll();	
    }

usermoko, 2021-03-04:
I don't know programming, but with your guidance i figured it out.
Thank you very much for your help.

closed 2021-03-04 by usermoko