karawin/Ka-Radio32 #212: Choose the ssid with best rssi when connecting to the network

menuet88, 2021-03-08:
I think this could be helpful if there are few networks with the same ssid but on different channels and rssi. For example when you walk with your radio around the house with few access points.
Now the Ka-Radio32 trying to connect last connected ap even when the signal is very week.

So my suggestion is to change  wifi_config in app_main.c at [line 535]( https://github.com/karawin/Ka-Radio32/blob/d7487a4501d5993ee4640685c8e5831c79e5f77b/main/app_main.c#L535) from:
```
wifi_config_t wifi_config = {
.sta = {
	.bssid_set = 0,
	},
};
```
to: 
```
wifi_config_t wifi_config = {
.sta = {
	.bssid_set = 0,
	.scan_method = WIFI_ALL_CHANNEL_SCAN,
	.sort_method = WIFI_CONNECT_AP_BY_SIGNAL,
       },
};
```

Of course this will slow down the connecting procedure, but I think the 2 seconds doesn't do big difference.

Best regards,
Menuet88

karawin, 2021-03-09:
ok, in the next release.

menuet88, 2021-03-09:
Great, thank you for accepting this.

closed 2021-03-29 by karawin