karawin/Ka-Radio32 #213: telnet: cli.url

oroszgergely, 2021-03-08:
hi there,

If im trying to play any station url via telnet by issuing the cli.url("stationUrl"),
 it confirms by echoing: ##CLI.URLSET#: stationUrl,
but the station wont start.
issuing the cli.start command starts the previously or currently played station, but wont switch to the new stationUrl one.

The above commands worked perfectly on prev rls-s.

Thanks in advance

karawin, 2021-03-08:
ou need to specify  three fields
##CLI.URLSET#: http://streaming.hotmix-radio.net
##CLI.PATHSET#: /hotmixradio-lounge-128.mp3
##CLI.PORTSET#: 80

Yes, i know, it is historical reason and must be improved.

oroszgergely, 2021-03-09:
thanks for your answer.
the cli.urlset worked some release ago, i dont remember if it was 1.9 or some older 2.0, but it worked, i used it in a nodejs script.
i tried the commands you suggested:

Karadio telnet
              > cli.urlset("http://streaming.hotmix-radio.net")
##CLI.URLSET#: http://streaming.hotmix-radio.net
                                                > cli.pathset("/hotmixradio-lounge-128.mp3")
##CLI.PATHSET#: /hotmixradio-lounge-128.mp3
                                           > cli.portset("80")
##CLI.PORTSET#: 80
                  > cli.start
///here happens what i said, it starts to play the previously played station, doesnt matter, what i entered with cli commands
##CLI.NAMESET#: 0 Rádio 1
##CLI.URLSET#: http://stream2.radio1.hu
##CLI.PATHSET#: /high.mp3.m3u
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 254
 ##CLI.URLSET#: http://stream2.radio1.hu
##CLI.PATHSET#: /high.mp3
 ##CLI.PORTSET#: 80
 ##CLI.STOPPED# from C_PLIST
 ##CLI.ICY0#: Radio 1
 ##CLI.ICY3#: http:\/\/www.radio1.hu
 ##CLI.ICY4#: pop
##CLI.ICY5#: 320
#CLI.ICY6#: Csak igazi mai slager megy
 ##CLI.PLAYING#

karawin, 2021-03-09:
In fact the right command is 
cli.instant

and not 
cli.play

I forgotten this point. ;-)

closed 2022-02-06 by karawin