karawin/Ka-Radio32 #215: PCM5102

Zinci, 2021-03-22:
I have one VS1053 board and this works fine with ESP32 and ESP8266. I also have a PCM5102 and that does not work. I connected the PCM5102 according to the schematic diagram:

PCM5102 -  ESP32
VIN           -  5V
GND        -  GND
LRCK       -  D25
DIN         -  D22
BCK         -  D26

I also tried the SCK to 5V through a resistor, but no joy.  
The Settings on a Web page was set to I2S.
Does this board work with ESP32 or did I do something (probably) wrong?

karawin, 2021-03-29:
Please read https://github.com/karawin/Ka-Radio32/blob/master/HardwareConfig.md
It depends on the configuration file. But you are right, the default is standard_adb.csv
To connect it do:
FLT GND
DMP GND
SCL Not connected
BCK is P_I2S_BCLK
DIN is P_I2S_DATA
LCK is P_I2S_LRCK
FMT GND
XMT 3.3V
VCC 5V or
3.3 3.3v
GND GND

Zinci, 2021-03-29:
Thanks, I completely missed the files on main page. Sorry about that. I usually read everything first. My bad.