karawin/Ka-Radio32 #217: Incorrect output of metadata

Pavel-sh123, 2021-04-05:
Hello, when assembling a receiver with UART control, faced with the following problem. When listening to Adiostania Radio Vanya, on the display is displayed instead of Russian characters here is this sequence & # 1044; & # 1048; & # 1057; etc. At first, I thought this error in my code, but no on the Karadio display the same thing, and on the web page, all inev. I would be very grateful for the help. 
[image]

karawin, 2021-04-15:
It is an unicode set of characters. 
Not decoded by KaRadio. 
Sorry.

Pavel-sh123, 2021-04-22:
It is clear, I will try to write a script on the control side for decryption.

Aleksey-748, 2021-11-02:
I managed to write a small procedure for converting characters in the format string & # 1040; in the format 0xD0A0, if possible, add it to the project.
//------------------------------------------------------------------------------//
void Cyrillic_decoding(void){                                                   //
union {                                                                         // 
    struct {                                                                    //
        unsigned char  t_LOW;                                                   //
        unsigned char  t_HIG;                                                   //
    };                                                                          //
    uint16_t T_t;                                                               //
}T_t;                                                                           //
uint16_t ss = 0;
uint16_t sss = 0;
size_t len = 0;
uint8_t l = 0;

    while (strstr(string,"&#") != NULL){
        
        len = strcspn(string, "&#");
        
        if(len == 0){l=1;}
        else{    
            for(uint16_t s=sss;s<len;s++){
               string_rec[ss++] = string[s];
               sss++;
            } 
        }
        string[len++] = ' ';
        string[len++] = ' ';

        uint8_t p_s[4];
        for(uint8_t a=0;a<4;a++){
           p_s[a] = string[len++];
        }
        
        T_t.T_t = atol(p_s);
        
        T_t.t_HIG = T_t.t_HIG << 2;
        T_t.t_HIG = (T_t.t_HIG & 0x3F) ^ (  T_t.t_LOW & 0xC0 );
        T_t.t_LOW = T_t.t_LOW & 0x3F ^ 0x80;
        T_t.t_HIG = T_t.t_HIG & 0x1F ^ 0xC0;
        
        string_rec[ss++] = T_t.t_HIG;
        string_rec[ss++] = T_t.t_LOW;
        
        sss = sss + 7;
        if(len>=  sizeof(string)){break;}
    }

    if(len < sizeof(string)){
        l = 1;
    }

    if(l){
        len =  sizeof(string);
        for(uint16_t s=sss;s<len;s++){
            string_rec[ss++] = string[s];
            sss++;
        }
    }

}
//------------------------------------------------------------------------------

karawin, 2021-11-06:
Thanks I will try to update soon.

karawin, 2022-01-18:
I took some time to integrate your function (for color lcd only for the moment.
It works now and be delivered in the next release.
Just changed all sizeof(string); by strlen(string) and some minor changes to make it working.
If you want to test it just OTA the bin with the command
sys.prerelease

Aleksey-748, 2022-01-18:
OK, thanks, with this function, a decent amount of SRAM memory will be freed up in my project.

karawin, 2022-01-20:
Update again with the sys.prerelease

Aleksey-748, 2022-01-20:
It got better, but still sometimes incorrect data slips, it seems that either the original line does not receive all the data, or something is cut off after the decoder, in general, the problem arises with long names.

Aleksey-748, 2022-01-20:
At home, I had to increase the volume of the input line to 897 bytes, for the decoder to work correctly.

karawin, 2022-01-20:
#At home, I had to increase the volume of the input line to 897 bytes, for the decoder to work correctly. #

I don't understand. Which input line, where?

Aleksey-748, 2022-01-21:
My project involves managing and receiving KAradio metadata via uart. So, the method data straka to which the yuart data comes originally had a size of 128 bytes or 64 characters in a two-byte encoding. But since the encoding is like "&#1044;" assumes 7 bytes per character, then the method data string had to be increased by 7 times, respectively, it became 896 + 1 bytes per 64 characters.

karawin, 2022-01-22:
I changed the serial output. Now the decoding is done before the output to the serial output.
It must be easier for your project.
If you agree, you can close the issue.

Aleksey-748, 2022-01-22:
Thank you so much, yes, the topic can probably be closed, now everything works as it should, the output on the KAradio display is correct, as well as on uart.

closed 2022-02-06 by karawin