karawin/Ka-Radio32 #219: LCD ST7920 shows only rubbisch data after starting music

ArnLil, 2021-05-09:
Hi, i have the a problem with ST7920, i am on the latest version 2.1. For me is this a new install set-up.
I also use a VS1053 for encoding the data. When i choose DAC for example, the display is working fine. So clould there be a problem witch the combination of ST7920 and the VS1053 ?
After starting the music (Sounds good) the display only shows rubbisch data.

What can i do to make it work ?  Thankx Arnold.

karawin, 2021-05-13:
May be the same CS for both devices. 
Check the gpio for each CS.

ArnLil, 2021-05-15:
The CS is not the same, i used the standard assignments for VS1053 and LCD. The only shared GPIO's are 23 KSPI MOSI (R/W on LCD) and 18 KSPI CLK.  For CS on LCD i used GPIO 13, for the VS1053 GPIO 32 is used.

karawin, 2021-05-16:
What file is flashed at 0x3A2000 ?
It seems to be an old problem with this lcd
https://www.eevblog.com/forum/projects/how-to-properly-connect-lcd-128x64-via-spi/?fbclid=IwAR1qXZ1PZc4aHiroQldfHpb65_blHlXTpZFNmsPrjJWYfHn-_qyr6dGr89Y

ArnLil, 2021-05-18:
Hi Karawin, your link brought me to the solution, i used the diode and resistor methode from https://github.com/olikraus/u8g2/issues/367#event-1272813635 the display is now fine !  btw i used the standard_adbnoled bin on 0x3A2000.
Thank you ! problem solved !

closed 2021-05-18 by ArnLil

karawin, 2021-05-18:
Great, thanks for the feedback.