karawin/Ka-Radio32 #223: No sound after changing station

bpardo, 2021-06-03:
After listening to a stream for several minutes, the change of station does not work correctly, the display shows the stream information correctly, but no sound comes out of the VS1053. IMHO a reset of the VS1053 at the station change would fix this problem.

##CLI.ICY0#: Duna-Tisza R�di� - Gold
##CLI.ICY3#: https:\/\/dunatiszaradio.hu
##CLI.ICY4#: Various
##CLI.ICY5#: 192
##CLI.ICY6#: A vid�k r�di�ja
##CLI.ICY7#:  ice-samplerate=44100;ice-bitrate=192;ice-channels=2
##CLI.META#: Europe - The Final Countdown
##CLI.PLAYING#
E (425075) vs1053: VS1053 Reset
##CLI.NAMESET#: 20 Antena 1 (BRAZIL)
##CLI.URLSET#: http://23.111.186.218
##CLI.PATHSET#: /stream/1
##CLI.PORTSET#: 8232
##CLI.OVOLSET#: 0
##CLI.VOL#: 141
E (428395) webclient: Socket: 60  connect errno:113 
##CLI.ICY0#: Connection error
##CLI.STOPPED# from Invalid

cheise, 2021-07-05:
I have tried out the VS1053 for a long time with Karadio and other solutions, but the results are all very buggy. 

Best solution for me is a PCM5102 DAC Board with a ESP32 WROVER. Thats playing all streams 24/7 with no problems. Both Codec's, mp3 and AAC. 

Maybe this is a better way for you?

bpardo, 2021-07-08:
Although I'm new to esp32 I have a long experience as a professional developer and from my humble analysis I don't think it's the VS1053 (but I could be wrong) the radio can work 24/7 the problem arises when I change the radio, either through the rotary knob or the web interface. No sound comes out anymore. If I change after a few seconds it works, but if I leave it on for ten minutes or so, then the station change is no longer possible.

maybe send a reset to vs1053 to correct this

Hope this will be fix in the next release :)

karawin, 2022-01-18:
E (428395) webclient: Socket: 60 connect errno:113
##CLI.ICY0#: Connection error
##CLI.STOPPED# from Invalid

Nothing related to the vs1053, but with the network.