karawin/Ka-Radio32 #225: Rotary Encoder/Display

andrewbpowers, 2021-07-05:
First of all, thank you so much for this great project!

1.) I find the time in which you can set the radio station after holding down the rotary encoder for a long time quite short. Is there any way to change this?

2.) Is it possible to change the information and the way it is displayed on the display? The amount of information on a small OLED is a bit much and too small (for me ;-)).

Thanks in advance for any help and tips!
:-)

Take care...

emax73, 2024-03-27:
I using double encoder - Voluime/Station all very good now

ruslanin, 2024-03-27:
My encoder placed inside radio box and only part of special printed knob handler visible outside. It is very uncomfortable press and rotate knob in my version. I want to change stations only rotate knob not pressing on it
[image]