karawin/Ka-Radio32 #228: Question about Touch TFT

WillVi1, 2021-07-28:
In the features List stands: 

**Touch screen if the hardware supports it.

I ordered now this Banggood TFT: 
 2.8 Inch ILI9341 240x320 SPI TFT LCD Display Touch Panel SPI Serial Port Module, Brand: Geekcreit

The connection of the TFT itself is no problem. 
But, how do I connect the Touch hardware an how has touch to be configured?

cheise, 2021-08-03:
TOUCH SCREEN

P_TOUCH_CS of the t_cs pin of the touch or 255 if no screen.
Other pins are t_clk, t_din, t_do respectively the spi clk, mosi, miso. T_irq is not used.
The screen must be calibrated with the command sys.cali[brate]
The screen is divided in 5 zones:

    Center: Start/Stop
    Top: Volume+
    Bottom: Volume-
    Left: Station-
    Bottom: Station+
see in [https://github.com/karawin/Ka-Radio32/blob/master/HardwareConfig.md](url)

kebibg, 2024-03-07:
Hi,
I have another board where the touch uses CLK, MOSI, and MISO on different pins as an LCD.
When I see the code for LCD and Touch is use one command to initialize the SPI but I need to be on different ones.
Are that is possible?

Regards