karawin/Ka-Radio32 #237: In my media center project, I added a Cyrillic encoder of the format & # 1040; in the format 0xD0A0, if possible, add it to the caradio project.

Aleksey-748, 2021-11-02:
I managed to write a small procedure for converting characters in the format string & # 1040; in the format 0xD0A0, if possible, add it to the project.
//------------------------------------------------------------------------------//
void Cyrillic_decoding(void){ //
union { //
struct { //
unsigned char t_LOW; //
unsigned char t_HIG; //
}; //
uint16_t T_t; //
}T_t; //
uint16_t ss = 0;
uint16_t sss = 0;
size_t len = 0;
uint8_t l = 0;

while (strstr(string,"&#") != NULL){
    
    len = strcspn(string, "&#");
    
    if(len == 0){l=1;}
    else{    
        for(uint16_t s=sss;s<len;s++){
           string_rec[ss++] = string[s];
           sss++;
        } 
    }
    string[len++] = ' ';
    string[len++] = ' ';

    uint8_t p_s[4];
    for(uint8_t a=0;a<4;a++){
       p_s[a] = string[len++];
    }
    
    T_t.T_t = atol(p_s);
    
    T_t.t_HIG = T_t.t_HIG << 2;
    T_t.t_HIG = (T_t.t_HIG & 0x3F) ^ (  T_t.t_LOW & 0xC0 );
    T_t.t_LOW = T_t.t_LOW & 0x3F ^ 0x80;
    T_t.t_HIG = T_t.t_HIG & 0x1F ^ 0xC0;
    
    string_rec[ss++] = T_t.t_HIG;
    string_rec[ss++] = T_t.t_LOW;
    
    sss = sss + 7;
    if(len>=  sizeof(string)){break;}
}

if(len < sizeof(string)){
    l = 1;
}

if(l){
    len =  sizeof(string);
    for(uint16_t s=sss;s<len;s++){
        string_rec[ss++] = string[s];
        sss++;
    }
}
}
//------------------------------------------------------------------------------

karawin, 2022-01-18:
I took some time to integrate your function (for color lcd only for the moment.
It works now and be delivered in the next release.
Just changed all sizeof(string); by strlen(string) and some minor changes to make it working.
If you want to test it just OTA the bin with the command
sys.prerelease

Aleksey-748, 2022-01-18:
OK, thanks, with this function, a decent amount of SRAM memory will be freed up in my project.

closed 2022-02-06 by karawin