karawin/Ka-Radio32 #242: "Server low mem" when playing AAC stream using S/PDIF.

Ant-12, 2021-12-06:
As you know, S/PDIF output doesn't work on current firmware(V2.1 R2).
The problem is in the presence of [this file](https://github.com/karawin/Ka-Radio32/blob/master/components/audio_renderer/audio_renderer1.c).  The legacy build system based on GNU Make compiles every *.c file...
I had to build the firmware by myself using esp32_win32_msys2_environment_and_toolchain-20181001 and esp-idf-v3.3.5.
However, I encountered the problem of lack of RAM when playing AAC streams via S/PDIF output on Wrover-B.
This happens when opening or updating the web interface page.

karawin, 2021-12-18:
I can't test S/PDIF output.
What is wrong and must be changed in the code?

Ant-12, 2021-12-25:
> What is wrong and must be changed in the code?

For what?
To enable S/PDIF output you need to rename or remove [the file](https://github.com/karawin/Ka-Radio32/blob/master/components/audio_renderer/audio_renderer1.c) specified above.
This is because "\components\audio_renderer\" folder now contains two files: "audio_renderer.c" and "audio_renderer1.c".
Another way is switching to CMake based build system.
I don't have solution for "Server low mem". Probably the webserver should use PSRAM instead of internal RAM.

karawin, 2022-02-01:
sys.prerelease to test the pre 2.2R1 please.

Ant-12, 2022-02-01:
2.2R1 - bug still present(I was trying to update web page at 71624 ms):

##CLI.NAMESET#: 6 Oldies AAC
##CLI.URLSET#: http://www.easyrockparadise.com
##CLI.PATHSET#: /
##CLI.PORTSET#: 8400
##CLI.OVOLSET#: 0
##CLI.VOL#: 254
D (50954) renderer: Renderer volume max:  65536  10000
I (50954) webserver: playstationInt: 6, g_device: 0
D (50954) eeprom: saveDeviceSettings
I (51154) webclient: watermark : 688  1672
D (51154) webclient: Socket: 62
I (51154) webclient: IP: bba6aa6b   ADDR:107.170.166.187
D (51344) webclient:
Sent: GET / HTTP/1.1
Host: www.easyrockparadise.com
icy-metadata: 1
User-Agent: karadio32

D (51554) webclient: contentType: 3
##CLI.ICY0#: Oldies Paradise
##CLI.ICY1#: <BR>This stream requires <a href=\"http:\/\/www.winamp.com\">Winamp<\/a><BR>
##CLI.ICY2#: SHOUTcast DNAS\/posix(linux x64) v2.5.1.724<BR>
##CLI.ICY3#: http:\/\/www.oldiesparadise.com
##CLI.ICY4#: Oldies, 60s, 70s, Pop, Rock
##CLI.ICY5#: 48
D (51614) webclient: t1: 0x3ffc38d3, cstatus: 2, icyfound: 1  metad:16384 Metaint:16384

D (51614) renderer: Renderer volume:  0
D (51614) renderer: Start
D (51614) webclient: newlen: 0   len: 515   chunked:0  pdata:3ffc36d4
D (52204) webclient: clientReceiveCallback: pdata: 3ffc36d4, pdataend: 3ffc3c74, len: 1440
D (52204) webclient: mt len:1440, clen:1440, metad:364 ,&l:3ffc3840, l:48, inpdata:3ffc36d4, rest:1027, str: StreamTitle='GREATEST HITS USA - HOUR 3 PART1';
##CLI.META#: GREATEST HITS USA - HOUR 3 PART1
##CLI.PLAYING#
D (53244) renderer: Renderer volume max:  65536  10000
D (55744) audio_player: RAM left 3871372
D (55744) audio_player: decoder task created: fdkaac_decoder_task
I (55744) audio_player: Buffer fill 16%, 65536 // 409600 bytes
W (55744) renderer: chip revision 1, enabling APLL
D (55774) renderer: changing sample rate from 0 to 44100
I (55834) fdkaac_decoder: pcm_size 8192, channels: 2, sample rate: 44100, object type: 2, bitrate: 60809
D (57064) webserver: watermark serverClientTask: 11c  284
D (71624) webserver: Heap size: 3586588
E (73664) servers: Server low mem. Retrying...
E (75664) servers: Server low mem. Retrying...
E (77664) servers: Server low mem. Retrying...
D (77724) webserver: watermark serverClientTask: 5fc  1532
E (79664) servers: Server low mem. Retrying...
D (79664) webserver: Heap size: 3586144
I (81444) audio_player: Buffer fill 14%, 58176 // 409600 bytes
E (81674) servers: Server low mem. Retrying...
E (83674) servers: Server low mem. Retrying...
E (85674) servers: Server low mem. Retrying...
D (85704) webserver: watermark serverClientTask: 648  1608
E (87674) servers: Server low mem. Retrying...
D (87674) webserver: Heap size: 3620840
D (87774) webserver: Heap size: 3588016
E (89784) servers: Server low mem. Retrying...
E (91784) servers: Server low mem. Retrying...
E (93784) servers: Server low mem. Retrying...
D (93824) webserver: watermark serverClientTask: 648  1608
E (95784) servers: Server low mem. Retrying...
D (95794) webserver: Heap size: 3618476
D (95794) webserver: HandlePost /icy

D (95794) webserver: icy start header 3ffc3690  len:405 vollen:3 vol:254
D (95824) webserver: Heap size: 3585704
D (95824) webserver: HandlePost /wifi

D (95854) webserver: Heap size: 3618476
D (95854) webserver: HandlePost /hardware

D (95874) webserver: Heap size: 3585704
D (95874) webserver: HandlePost /rauto

D (95904) webserver: Heap size: 3585704
D (95904) webserver: HandlePost /theme

E (97794) servers: Server low mem. Retrying...
E (99794) servers: Server low mem. Retrying...
E (101794) servers: Server low mem. Retrying...
D (101914) webserver: watermark serverClientTask: 1cc  460
E (103794) servers: Server low mem. Retrying...
D (103794) webserver: Heap size: 3620420
D (103804) webserver: websocketAccept socket: 61
D (103804) webserver: watermark serverClientTask: 60c  1548
I (107784) audio_player: Buffer fill 13%, 56907 // 409600 bytes