karawin/Ka-Radio32 #243: Ka-Radio32/components/freertos/include/freertos is missinig

pierre04, 2021-12-12:
Hello. I am trying to build this project, and face an issue when setting esp32 target :
`~/esp/Ka-Radio32$ idf.py set-target esp32`

lead to the error 

`CMake Error at /home/pierre/esp/esp-idf/tools/cmake/component.cmake:305 (message):
  Include directory
  '/home/pierre/esp/Ka-Radio32/components/freertos/include/freertos' is not a
  directory.
Call Stack (most recent call first):
  /home/pierre/esp/esp-idf/tools/cmake/component.cmake:477 (__component_add_include_dirs)
  /home/pierre/esp/esp-idf/tools/cmake/component.cmake:538 (idf_component_register)
  components/wolfssl/CMakeLists.txt:37 (register_component)`

The esp32 idf is freshly installed on a Linux Box. I can successfully compile and flash the exemples «hello world» and «blink»
[CMakeOutput.log](https://github.com/karawin/Ka-Radio32/files/7698846/CMakeOutput.log)

I attach the compilation log.
[CMakeOutput.log](https://github.com/karawin/Ka-Radio32/files/7698846/CMakeOutput.log)

Ant-12, 2021-12-13:
KA-RADIO32 is incompatible with ESP-IDF V4 and newer. It uses gnu make instead of cmake.
I successfully build it using _esp32_win32_msys2_environment_and_toolchain-20181001_ and _esp-idf-v3.3.5_.

Ant-12, 2021-12-13:
> Do you plan to support ESP-IDF V4 in the future ?

You better ask [contributors](https://github.com/karawin/Ka-Radio32/graphs/contributors).

Regarding me, I'm going to try it, but not sure that I have enough experience for success.
There is a [fork](https://github.com/pilnikov/Ka_mod) which is compatible with modern esp-idf but it's outdated little bit.

pierre04, 2021-12-13:
Oh, I tought you were part of the project :)
I'll look at the fork. 
It seems that all those nice projects seems to be abandoned. I wonder if I should not use a much simpler arduino sketch.

karawin, 2022-03-16:
See also the version for ESP-IDF v4.4 at
https://github.com/karawin/KaRadio32_4

closed 2022-03-16 by karawin