karawin/Ka-Radio32 #244: Changing to another LCD

CannaW, 2021-12-16:
I'm sorry, but I'm not familiar with programming and compilation.
I read in various files that I can change LCD after changing the sys.lcd("x") parameter in the HardwareConfig.md file. In another place I read that a .csv or .bin file has to be generated. I even zaisntalled Eclipse, esp-idf-v4.0.4 and MSYS64bit. However, I do not understand much of this.
Can I count on a hint?

karawin, 2021-12-18:
sys.lcd("x")  is the command to change the lcd type.
It must be done one time only and registered in the configuration.
This command can be sent with telnet or the serial I/O

CannaW, 2022-01-23:
Big thanks for the tip. I works great. Now I can work on the nice cover. I'll going to send to you final effect. Great work!

closed 2022-02-06 by karawin