karawin/Ka-Radio32 #248: app_main.c error: cannot convert to a pointer type

h1aji, 2021-12-28:
I am seeing this error message when I am trying to compile the code with esp-idf 3.3.1
```
esp/Ka-Radio32/main/app_main.c:672:3: error: cannot convert to a pointer type
   ip_addr_t *ipdns0 = (ip_addr_t *)dns_getserver(0);
```
Is this piece of code supposed to work?
https://github.com/karawin/Ka-Radio32/blob/571d4c9cfc7de54758d13e2592883516c9008c3d/main/app_main.c#L672

I had to reverse it and set it to 
`		ip_addr_t ipdns0 = dns_getserver(0);
`
to make working

Ant-12, 2021-12-28:
Try esp-idf 3.3.5

h1aji, 2021-12-29:
Thank you @Ant-12 it worked.

@karawin would you be able to update README.md and set correct sdk version?
I have created PR. Thanks

closed 2021-12-30 by h1aji