karawin/Ka-Radio32 #250: not play stream radio garden

electro-nic, 2022-01-15:
I try play link https://stream.radiojar.com/8du64725e3quv?listening-from-radio-garden=1642243580 but not to do
What is wrong?
VLC play this link

karawin, 2022-01-17:
It play

##CLI.URLSET#: https://stream.radiojar.com
##CLI.PATHSET#: /8du64725e3quv?listening-from-radio-garden=1642243580
##CLI.PORTSET#: 443
##CLI.OVOLSET#: 0
##CLI.VOL#: 47
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: https://n0b.radiojar.com
##CLI.PATHSET#: /8du64725e3quv?listening-from-radio-garden=1642243580&rj-tok=AAABfmnTsAIAWaLorefQo9mP3w&rj-ttl=5
##CLI.PORTSET#: 443
##CLI.ICY0#:  
##CLI.PLAYING#

##CLI.META#: Î”Î—ÎœÎ—Î¤Î¡Î‘ Î“Î‘Î›Î‘ÎÎ— - Î¤ÎŸ Î”Î™Î§Î¤Î¥ (Î–)

karawin, 2022-01-18:
You can also try http://stream.radiojar.com/8du64725e3quv?listening-from-radio-garden=1642243580
##CLI.URLSET#: http://stream.radiojar.com
##CLI.PATHSET#: /8du64725e3quv?listening-from-radio-garden=1642243580
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
##CLI.VOL#: 40
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: http://n09.radiojar.com
##CLI.PATHSET#: /8du64725e3quv?listening-from-radio-garden=1642243580&rj-tok=AAABfm257R4ACR9jhwWVr2WxJg&rj-ttl=5
##CLI.PORTSET#: 80
##CLI.ICY0#:  
##CLI.META#: Î£Î¤Î‘Î¥Î¡ÎŸÎ£ Î£Î™ÎŸÎ›Î‘Î£ & Î¡Î•ÎÎ‘ ÎœÎŸÎ¡Î¦Î— - Î£Î‘Î Î§ÎŸÎ¥Î‘Î
##CLI.PLAYING#

closed 2022-01-18 by karawin